hzeller/rpi-rgb-led-matrix #503: 8x16,16x16 Matrix

AraiHiroaki, 2018-02-02:
Hello @hzeller,thank you for your wonderful libraly.I have 16 x 32, 32 x 32, HUB 75 LED Matrix control is successful.
However, I'd like to control outdoor LED Matrix HUB75,P16 or P20.They are 16x16 or 8x16 matrix.
I tried "--led-multiplexing",but it is not going well. I would be pleased if you could advise me.

hzeller, 2018-02-02:
So, if it is not going well, how does it look instead ?

AraiHiroaki, 2018-02-02:
Thank you for your reply.

I have two 16x16(P16) LED matrix panels.
I connected them.

[image]

./demo -D0 -r 16 -c 2

[image]

./demo -D0 -r 16 -c 4

[image]

./demo -D0 -r 16 -c 8

[image]

./demo -D0 -r 16 -c 16

[image]

AraiHiroaki, 2018-02-02:
./demo -D 0 -r32 -c8 --led-multiplexing 0
https://vimeo.com/253975708

./demo -D 0 -r32 -c8 --led-multiplexing 1
https://vimeo.com/253975580

./demo -D 0 -r32 -c8 --led-multiplexing 2
https://vimeo.com/253975595

./demo -D 0 -r32 -c8 --led-multiplexing 3
https://vimeo.com/253975799

hzeller, 2018-02-02:
Can you take pictures with the `-D3` demo ? It might be easier to see what is going on as this just shows a static square.

Why are you choosing -r32, shouldn't this be -r16 as these are panels with 16 rows ?
Also why did you choose a led-chain of 8 (`-c3`) ? In the picture above you only have two panels.

So testing with various led multiplexing of
  `./demo -D3 --led-rows=16 --led-cols=16 --led-chain=2 --led-multiplexing=<0,1,2,3>`
and taking pictures would be good.
Also the same test, but with only `--led-chain=1` and then a picture of the first panel in the chain.

It looks like it is mostly ok but then mirrored in the middle.

How does the 8x16 panel look like you mention. Is it 8 high and 16 long or 16 high and 8 long ? Can you send a picture of the back of the 8x16 panel in a way that the input connector and potential printings next to it are clearly visible ?

AraiHiroaki, 2018-02-03:
Thank you for your instruction.

For -r32 - c8,because I seemed to be doing better.(I tried many combinations.)
ex) ./demo -D 0 --led-rows=16  --led-cols=16 --led-chain=2 --led-multiplexing 0
[image]

But now,I'll take your advice.

./demo -D 3 --led-rows=16  --led-cols=16 --led-chain=2 --led-multiplexing 0
[image]

./demo -D 3 --led-rows=16  --led-cols=16 --led-chain=2 --led-multiplexing 1
[image]

./demo -D 3 --led-rows=16  --led-cols=16 --led-chain=2 --led-multiplexing 2
[image]

./demo -D 3 --led-rows=16  --led-cols=16 --led-chain=2 --led-multiplexing 3
[image]

I try the same with a single panel.

./demo -D 3 --led-rows=16  --led-cols=16 --led-chain=1 --led-multiplexing 0
[image]

./demo -D 3 --led-rows=16  --led-cols=16  --led-chain=1 --led-multiplexing 1
[image]

./demo -D 3 --led-rows=16  --led-cols=16 --led-chain=1 --led-multiplexing 2
[image]

./demo -D 3 --led-rows=16  --led-cols=16 --led-chain=1 --led-multiplexing 3
[image]

AraiHiroaki, 2018-02-03:
And the 8x16 panel is as following pictures.

[image]
[image]
[image]
[image]
[image]

angelogoncalve, 2018-02-04:
See this issue https://github.com/hzeller/rpi-rgb-led-matrix/issues/136.

hzeller, 2018-02-17:
So if all the multiplex options [0..4] don't work, you might need to provide the mapping that works for your panel and send a pull request so that we can include it in the toplevel. You essentially need to add code to [lib/multiplex-transformers.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-transformers.cc) and need to wire it up [around here](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L193).

AraiHiroaki, 2018-02-18:
Thanks hzeller.
I read issue # 136. over and over,and it improved.

sudo ./demo -D 2 runtext.ppm -L --led-parallel=2 --led-rows=8 --led-chain=16 --led-cols=16
Size: 128x32. Hardware gpio mapping: regular
Read image 'runtext.ppm' with 500x32
Press <CTRL-C> to exit and reset LEDs

[image]

angelogoncalve, 2018-02-18:
Are you used any transformer Dear AraiHiroaki?

AraiHiroaki, 2018-02-20:
Thank you, angelogoncalve. 
I used the following formula.
(It is not complete yet....)

new_x=(1-(x/4 % 2))*64+(x % 4)*4+(y % 4)+(y/4 % 4)*16+(y/16 % 2)*128;
new_y=x % 4 + 4*((x / 8));

angelogoncalve, 2018-02-20:
Thanks my friend when you put the demo working with your finished formula (transformer) for the 16x16 panel, please share for the community. Thank you very much for your help Dear friend.

AraiHiroaki, 2018-03-04:
Thanks dear angelogoncalve.
I repeated the test many times, but I can't make much progress.

With the following transformer,

new_x=4*((1-(x/4 % 2))*16+(x % 4)+(y/4 % 4)*4+(y/16 % 2)*32)+(y % 4);
new_y=x % 4 + 4*((x / 8));  

sudo ./demo -D 1 runtext.ppm --led-parallel=2 --led-rows=8 --led-chain=8 --led-cols=32

https://vimeo.com/258464736

Because of my lack of understanding about lib,I don't know why "--led-rows=8 --led-chain=8 --led-cols=32".This Panel is 16x16 matrix.
I think correctly "--led-rows=16 --led-chain=2 --led-cols=16". (??)

angelogoncalve, 2018-03-04:
Yes the test is sudo ./demo -D 1 runtext.ppm --led-parallel=2 --led-rows=8 --led-chain=8 --led-cols=32 --led-pwm-bits=1 --led-slowdown-gpio=4