hzeller/rpi-rgb-led-matrix #504: Can GPIO18 be swapped to another pin

aussieW, 2018-02-04:
I would like to combine an LED matrix along with a DAC running on I2S. They both need to use GPIO18. Could I swap for another unused GPIO say GPIO20 within the rpi-rgb-led-matrix library? If so which files would I need to modify?

Cheers

hzeller, 2018-02-04:
You can use another Pin, _however_, then rpi-rgb-led-matrix can't use the timer hardware circuit in the Pi to generate the exact timed output enable timings. This means that you have sometimes noticeable brightness glitches of random single lines. It is the same effect as using `--led-no-hardware-pulse`.

Now I wonder why you need GPIO18 ? The I2C is on Pin 3 and 5 on the Pi ( https://pinout.xyz/pinout/i2c ) while the GPIO18 is on Pin12, so they shouldn't interfere. Or is it that the DAC also uses GPIO18 to generate exact hardware pulses ?

Anyway, you can choose a different GPIO in the [hardware-mapping.c](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c#L32) file and change a different GPIO for `output_enable`, now at `GPIO_BIT(18)`. Be aware that essentially all pins are needed for the three-chain configuration of the Pi, so you can't run three parallel chains anymore if you also want to use the I2C pins that map to GPIO 2 and 3 (the third chain uses these Pins, so you can go up to `--led-chain=2`). So I suggest to move the `output_enable` to one of the pins from chain3 in that case that you can't use anyway anymore (GPIO 14, 26, 16 or 21).

aussieW, 2018-02-04:
You are probably correct in your assumption about the DAC. It seems to use GPIO18 for the "Audio data bit clock input". Here is a web page that describes how to wire the DAC up to the pi (http://http://masterandrey.com/posts/en/dac_raspberry_pi_pcm5102/) . Neither of the DACs in the article matches my DAC module exactly, but it is based on the same chip PCM5102a.

I see GPIO18 is referenced several time in https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.cc. Do I need to change those as well?

aussieW, 2018-02-04:
Okay, thanks. I'll give it a go.

aussieW, 2018-02-04:
Seems to work for the C based demos, but not for my python code. Anything extra I need to do to make the python bindings work?

hzeller, 2018-02-04:
You need to clean, re-compile and re-install the python binding.

hzeller, 2018-02-04:
Did it work ?

aussieW, 2018-02-04:
Well, yes and no. Yes the LED Matrix and the DAC both work independently. However, they don't work together. As soon as the LED Matrix code is started the DAC cuts out and doesn't work again till a reboot, even if the LED Matrix code is no longer running.
The DAC is enabled in the /boot/config.txt file using
```
dtparam=i2s=on
dtoverlay=hifiberry-dac
dtoverlay=i2s-mmap
```
Any thoughts?

hzeller, 2018-02-04:
Can you caomment out the line that sets GPIO18 as input ?
  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.cc#L153
This is only needed to work around some particular Adafruit HAT issue, but it will change the mode of the GPIO pin, which can be problematic in your context.

aussieW, 2018-02-04:
Some good news! That seems to have fixed the problem when running the C demos, both the DAC and LED Matrix work together. However, the problem still remains with the python bindings. My program runs in python so I need those to work. I did follow the instructions to reinstall the python bindings.

hzeller, 2018-02-04:
So the Python thing is just a wrapper around the C++ library. If things are not working with the Python binding something has not picked up the change. Make sure to run a `sudo make clean` in the bindings/python directory first before doing the rebuilding.
Also verify that you have compiled it for the same Python version you are using; otherwise you might just use an old version of the bindings for a different Python version you compiled.

aussieW, 2018-02-04:
That worked! Thanks a lot for your help.

hzeller, 2018-02-04:
Cool. In the meantime, I've updated the code so that the part that you had commented out should now work automatically (https://github.com/hzeller/rpi-rgb-led-matrix/commit/d9ca28ad38f4670295aa748523955afbf79e7152).
So if you do a git pull, be aware that you might get a conflict in that line, but it should be easy to resolve.

closed 2018-02-04 by hzeller