hzeller/rpi-rgb-led-matrix #507: Weird multiplexing mode?

mansj, 2018-02-07:
I can't get any of the multiplexing modes to work right on my P10 16x32. 

See video: https://www.youtube.com/watch?v=ay-zYqXtmBA

This video ran `sudo ./examples-api-use/demo -D1 --led-gpio-mapping=adafruit-hat --led-cols=32 --led-rows=16  --led-brightness=40 examples-api-use/runtext16.ppm --led-multiplexing=3`

But all multiplexing modes look wrong in different ways. 

Any insight?

hzeller, 2018-02-07:
mmh, this looks like a tradiional 1:8 multiplexing but with a 'reflected' second stripe.
What does the output with the flag `-D3 --led-cols=32 --led-rows=16` look like ?

mansj, 2018-02-08:
Here's that mode with no other flags except adafruit-hat:

[image]

angelogoncalve, 2018-02-08:
Use sudo ./examples-api-use/demo -D1 --led-gpio-mapping=adafruit-hat --led-cols=32 --led-rows=16 --led-chain=1 --led-parallel=1 --led-brightness=100 examples-api-use/runtext16.ppm --led-multiplexing=1

mansj, 2018-02-09:
No good unfortunately, @angelogoncalve : 
[image]

hzeller, 2018-02-09:
So looks like you need a transformer for this that maps 32x16 to 32x16 but with every other stripe of 4 going from the backwards. There is no standard transformer for this yet compiled-in, but it should be failry straight-forward for you to write.

Since this is not a multiplexing that reduces the number of lines, the standard multiplexing set-up doesn't work for that yet, but hopefully I can get some time on the weekend to prepare it in a way that it is easier to do.

angelogoncalve, 2018-02-09:
Use sudo ./examples-api-use/demo -D1 --led-gpio-mapping=adafruit-hat --led-cols=16 --led-rows=8 --led-chain=2 --led-parallel=1 --led-brightness=100 examples-api-use/runtext16.ppm --led-multiplexing=1

angelogoncalve, 2018-02-09:
or this way sudo ./examples-api-use/demo -D1 --led-gpio-mapping=adafruit-hat --led-cols=16 --led-rows=8 --led-chain=2 --led-parallel=1 --led-brightness=100 examples-api-use/runtext16.ppm --led-multiplexing=1

angelogoncalve, 2018-02-11:
or use this way mode: --led-rows=16 --led-row-addr-type=2 --led-multiplexing=4

angelogoncalve, 2018-02-11:
or this one: --led-rows=16 --led-row-addr-type=2 --led-multiplexing=0

mansj, 2018-02-12:
Thanks, unfortunately during all this testing it seems that my (very cheap) LED panel has died, today I can't seem to get it to work at all, so most likely something burned out somewhere during my testing. I'll see if I can get a new one at the same price and try again.

hzeller, 2018-02-17:
Alright. Once you're there, check out all the multiplexing options (another one has been added last week). If it doesn't work, just send a pull request with the binding needed for your panel.

closed 2018-02-17 by hzeller