hzeller/rpi-rgb-led-matrix #511: Image Viewer make error

mike-rankin, 2018-02-10:
I have the latest Raspberri Pi os version with all the examples working perfectly. I hoped to try the Image Viewer but see errors when running make.

sudo apt-get update
sudo apt-get install libgraphicsmagick++-dev libwebp-dev -y
make led-image-viewer

g++ -I../include -Wall -O3 -g -Wextra -Wno-unused-parameter `GraphicsMagick++-config --cppflags --cxxflags` -c -o led-image-viewer.o led-image-viewer.cc
Assembler messages:
Fatal error: can't create led-image-viewer.o: Permission denied
Makefile:37: recipe for target 'led-image-viewer.o' failed
make: *** [led-image-viewer.o] Error 1

hzeller, 2018-02-11:
Did you run `make` as root before and now have a file that you can't write as regular user ? Or you checked out the directory as root ? This is probably the issue.
So just check it out afresh as regular user and compile as such and it will work.

mike-rankin, 2018-02-11:
You are correct. A simple sudo in front of make led-image-viewer works fine.

A huge thank you for all your work on this library! A massive amount of effort has gone into it.

closed 2018-02-17 by hzeller