hzeller/rpi-rgb-led-matrix #512: Help with bitmap file

Toootless, 2018-02-12:
I am working on a project were the raspberry will be hooked up and a module from a vehicle that has four ultrasonic sensors. I plan on using a 32x16 display to display what the sensors are detecting. I want to break the display into four groups and tie each group to the sensor output. the problem is I do not know how to go about this. I was thinking of having them grouped by 8 leds wide and then if a sensor detects something far you would see (starting in the lower left of the led matrix ) 3 rows tall by 8 columns wide in green. If the sensor saw it get closer then you would see it change to 7 rows by 8 columns wide turn yellow and then finally you would see 13 x 8 turn red.  Should I try to find a way to make an image of this type of output? or should I make a bdf file?

Toootless, 2018-02-12:
I may have found my own answer https://github.com/hzeller/rpi-matrix-pixelpusher.

hzeller, 2018-02-12:
The rpi-rgb-led-matrix provides a Canvas, where you can call SetPixel() at any position and color the way you want it, so it should be possible to just write your program directly in C++ or Python using this library.

If you want to go networked, like with the pixelpusher, you might also consider the flaschen-taschen protocol which is simpler and for which there is also a compile option to use this library: https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display

closed 2018-02-17 by hzeller