hzeller/rpi-rgb-led-matrix #513: example graphics.py only runs once

jo-elektro, 2018-02-12:
Hi, 

I have been playing a little with this excellent library in conjunction with a 64x32 RGB panel from Adafruit together with the clock HAT and focusing on the the python samples. 
On the software side I used the brand new installation script from Ada which was only updated last week and is pretty straight forward.
[](https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi/driving-matrices)

The sample work fine but I have run into a problem with the graphics.py example which only wants to run once in **Idle**. Running it a second time the terminal spits out a bunch of server call issues as listed bellow.
This happens only if I run the program from Idle as sudo. 
If I type sudo ./graphics.py --led-chain=2 everything is A-ok. 

----------------------------------------
Unhandled server exception!
Thread: SockThread
Client Address:  ('127.0.0.1', 49775)
Request:  <socket._socketobject object at 0x76c35d88>
Traceback (most recent call last):
  File "/usr/lib/python2.7/SocketServer.py", line 295, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 321, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python2.7/SocketServer.py", line 334, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python2.7/idlelib/rpc.py", line 503, in __init__
    SocketServer.BaseRequestHandler.__init__(self, sock, addr, svr)
  File "/usr/lib/python2.7/SocketServer.py", line 655, in __init__
    self.handle()
  File "/usr/lib/python2.7/idlelib/run.py", line 291, in handle
    rpc.RPCHandler.getresponse(self, myseq=None, wait=0.05)
  File "/usr/lib/python2.7/idlelib/rpc.py", line 280, in getresponse
    response = self._getresponse(myseq, wait)
  File "/usr/lib/python2.7/idlelib/rpc.py", line 300, in _getresponse
    response = self.pollresponse(myseq, wait)
  File "/usr/lib/python2.7/idlelib/rpc.py", line 424, in pollresponse
    message = self.pollmessage(wait)
  File "/usr/lib/python2.7/idlelib/rpc.py", line 376, in pollmessage
    packet = self.pollpacket(wait)
  File "/usr/lib/python2.7/idlelib/rpc.py", line 347, in pollpacket
    r, w, x = select.select([self.sock.fileno()], [], [], wait)
error: (4, 'Interrupted system call')

*** Unrecoverable, server exiting!
----------------------------------------

I have read a few thing about the no_block option but this is way past my pythonic comprehension and some help would be very appreciated. 

Greetings,

Jo

hzeller, 2018-02-17:
It is not clear what you are trying to do here. Why is your code having a socket connection open ?

Anyway, if you try to run two instances of programs in parallel that try to drive the hardware low-level, that won't work of course as they would both need the same resources. However, consecutively, they should work.

jo-elektro, 2018-02-19:
Hi,

I am not trying anything special I think. Just copied the example from the readme text in /home/pi/rpi-rgb-led-matrix/bindings/python into Idle and then try to execute it. 

#!/usr/bin/env python
import time
import sys

from rgbmatrix import RGBMatrix, RGBMatrixOptions
from PIL import Image

if len(sys.argv) < 2:
    sys.exit("Require an image argument")
else:
    image_file = sys.argv[1]

image = Image.open(image_file)

# Configuration for the matrix
options = RGBMatrixOptions()
options.rows = 32
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'regular'  # If you have an Adafruit HAT: 'adafruit-hat'

matrix = RGBMatrix(options = options)

# Make image fit our screen.
image.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)

matrix.SetImage(image.convert('RGB'))

try:
    print("Press CTRL-C to stop.")
    while True:
        time.sleep(100)
except KeyboardInterrupt:
    sys.exit(0)

jo-elektro, 2018-02-19:
I figured out that it trips on the line  **matrix = RGBMatrix(options = options** but I understand too little of this on how it is causing the program to open multiple instances.

closed 2025-11-27 by marcmerlin