hzeller/rpi-rgb-led-matrix #518: Re: Assistance for 10 minute countdown timer using real time clock and Raspberry pi

Arsenik4444, 2018-02-15:
Hi All, 
May be this not the right place for this topic but i need your a assistance and help regarding a School Project have been struggling for weeks:
It’s a 10 minute countdown timer with real time clock being displayed on four 5x7 LED boards; a short from a relay NO contact triggers it and an open circuit from the same contact would reset the timer back to 10 minutes awaiting the next trigger. Once the 10 minutes runs out after the contact is triggered we’ll remain on 0:00 until the contact goes open circuit again.
The communication between the board is done using Rs232 to Rs485(Uart communication between the LED board and the Raspberry pi) and the board will be addressed(1,2,3,4) ; please any assistance will be great;
Thank you to all.

Arsenik4444, 2018-02-15:
Hi All, 
If any possible to be implemented, we can do this with the 16x32 Rgb matrix by display(0:00) which is minute and Second;
Thank you all.

davemaster, 2018-02-15:
Greetings,

Make a GUI application, then capture it and show that on 16x32 rgb led
matrix.

hzeller, 2018-02-17:
You might want to start with the [examples-api-use/clock.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/clock.cc) example and adapt it to your needs.

closed 2025-11-27 by marcmerlin