hzeller/rpi-rgb-led-matrix #519: Node binding seems out of date, can't recompile either.

jcomack, 2018-02-15:
Hello,

First off I want to thank you for this lovely library and the great explanation on wiring up a panel to a Raspberry Pi!

I'm running into an issue with the Node binding over at  https://github.com/zeitungen/node-rpi-rgb-led-matrix/ where the current version seems to have a hard assertion that rows must be `<= 32` and I'm trying to run this with a 64 x 64 panel. As it seems that the library is no longer actively maintained, I attempted to update the `matrix` submodules, which resulted in some clear changes in the code. Awesome! However, after I forked said repository and attempted to push my 'fix' to my fork and pull it back in as a node dependency, I got hit with the following error message:

```
make: Entering directory '/home/pi/Code/Display/node_modules/node-rpi-rgb-led-matrix/build'
make: Leaving directory '/home/pi/Code/Display/node_modules/node-rpi-rgb-led-matrix/build'
make: *** No rule to make target 'Release/obj.target/rpi-rgb-led-matrix/external/matrix/lib/bdf-font.o', needed by 'Release/obj.target/rpi-rgb-led-matrix.a'.  Stop.
gyp ERR! build error
```

Now, my skills in writing Makefiles are... abysmal, to say the least. Could you help me to resolve this issue? Thanks.

jcomack, 2018-02-16:
Okay, seems like it was me running `yarn add` instead of `npm install` that caused this. However, now I'm getting `Error: /home/pi/Code/Display/node_modules/node-rpi-rgb-led-matrix/build/Release/node-rpi-rgb-led-matrix.node: undefined symbol: matrix_hardware_mappings` thrown at me.

Any tips on how I could resolve this?

hzeller, 2018-02-17:
w.r.t. the build issues with the node.js library: I don't know; it sounds like it is not using the original Makefile but redoes its own build system. Ideally, you'll send a PR there to use the original Makefile.
So if you know a thing about JavaScript binding, it might be also cool probably to send them a pull request to update to the latest version of the library and provide a binding to the `RGBMatrix::Options` struct.

In the C++ version of the library, you can either use the command line flags(`--led-rows=64 --led-cols=64 ...`), if you are using `CreateMatrixFromFlags()`, or you can set an [`RGBMatrix::Options`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/led-matrix.h#L56)  struct and call `CreateMatrixFromOptions()` (or combine the two; detailed [documentation around here in the header file](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/led-matrix.h#L388)).

If you want to lift the node.js binding to the latest version of this library here (and send them a pull request), I suggest to provide a binding around the `RGBMatrix::Options` struct to pass configuration options.

(Sometimes it is easier to do language bindings towards a C library, so there is also a [pure C header](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/led-matrix-c.h) you can use).

jcomack, 2018-02-17:
Thanks for the response. I'm gonna have to dig around a bit further first before making the PRs. Your pointers will certainly help :)

natcl, 2018-03-18:
Curious to see if there are any updates to this?

hzeller, 2018-03-18:
This is somewhat the wrong place, as the node.js binding is not maintained here; I'd ask on the project https://github.com/zeitungen/node-rpi-rgb-led-matrix/ directly and poke @zeitungen for update...