hzeller/rpi-rgb-led-matrix #520: Installation broke Python3 on Raspberry Pi 3

fedus, 2018-02-17:
Hello,

after following the tutorial on Adafruit using the install script (RPI3, Matrix HAT + RTC), my Python3 installation is completely broken.

When running Python3, I now get the following error message:
Fatal Python error: Failed to open?/dev/urandoi

Unfortunately, I couldn't make much sense of the Google results I've been getting for that message.

To make matters worse, trying to uninstall and reinstall Python3 using apt did not work and I now have tons of unmet dependencies, not to mention that some important system services based on Python3 are not working anymore.

Does anyone have an idea how this could be solved?
Many thanks.

hzeller, 2018-02-17:
Not sure why you are asking here about something that Adafruit writes. If you use an Adafruit tutorial, please ask them.

Probably the easiest is, to get a fresh image of Raspbian and then install the Python binding [according to the documentation in this library](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python).

fedus, 2018-02-17:
Apologies - I thought the install script used by Adafruit was part of this repository ... I now realised it is made by Adafruit.

Sorry for posting in the wrong forum.

closed 2018-02-17 by hzeller