hzeller/rpi-rgb-led-matrix #521: Panel P2.5 128x64 1/32 Hub75E

jtgijon, 2018-02-17:
Thanks for this unique library. I have the following setup:
- P2.5  128x64 1/32 panel with E line. This module seems to be new and it is not the typical combination of two 64x64 modules as it has a single hub75E input and a single hub75E output. I bought one from here: https://es.aliexpress.com/item/LYSONLED-P2-5-Indoor-SMD2121-Full-Color-Led-Display-Module-1-32-Scan-320x160mm-Super-High/32845686589.html?spm=a219c.10010108.1000016.1.1396373d3Eloob&isOrigTitle=true
- An active RPi matrix drive board from electrodragon ( #120 or #406)
- A Raspberry Pi3 with Raspbian Stretch Lite
- The Rpi-rgb-led-matrix downloaded today. Using Regular led GPIO mapping. Thanks again.

[image]

Hub75E is the following one:
[image]

It works the same using this two configurations:
`--led-rows=64 --led-cols=64 --led-chain=2` or 
`--led-rows=64 --led-cols=128 --led-chain=1`
...although it is needed to use `--led-slowdown-gpio=3` as shown in #120 (values not described in README.md yet)
The problem I have is similar than the issue commented by @AsafFisher in #120 , where there are displayed two strips of 16 leds each.
This is what I get when trying to execute demo D0
`sudo ./demo --led-rows=64 --led-cols=128 --led-chain=1 --led-slowdown-gpio=3 --led-brightness=10 -D 0`

[image]

`--led-multiplexing` change the multiplexing behaviour of the display but the black stripes continues.
`--led-row-addr-type=1` seems to do something, but is not adapted for this panel as shown in the following picture:
[image]

`sudo ./demo --led-rows=64 --led-cols=128 --led-chain=1 --led-slowdown-gpio=3 --led-brightness=10 -D 3`
[image]

I’ve tested 16 PIN IDC cable and it seems to work fine. 
Any help to solve this problem will be appreciated very much.

hzeller, 2018-02-17:
You have an E address line on Pin8 of the IDC.
Does your active matrix adapter have that connected [like the adapter provided in this project](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) ?

hzeller, 2018-02-17:
So if it is this board
http://s2.electrodragon.com/wp-content/uploads/2017/08/RGB-Matrix-Panel-Drive-Board-For-Raspberry-Pi-02.jpg
You need to make sure to connect the E-line accordingly.

jtgijon, 2018-02-18:
I forgot that connection. Now it works fine. Thanks,
[image]
[image]

hzeller, 2018-02-18:
cool, have fun with your matrix!

closed 2018-02-18 by hzeller

Nakamal, 2019-02-07:
Hi, we are working with almost identical equipment, we have a raspberry pi 3 B+, ada fruit RGB matrix bonnet for raspberry pi and the RGB Matrix Panel Drive Board For Raspberry Pi. We have 4 64x64 matrix boards but we are still just trying to make one of them display with the commands that we give. Do you have any advice (or hzeller if you're out there)? What sized batteries did you use?

jtgijon, 2019-02-07:
Hello. 4 64x64 in a U form should work.  I don't use adafruit bonnet but one from electrodragon (see avobe) and my pannel is a single module of 128x64 leds P2.5 1/32.  If you use adafruit I supposse you need to change at least --led-gpio-mapping=adafruit-hat
Please, describe the command line you use, the type of panel and what is displayed. There are some demo examples you can use to figure out what is happening (see "Running some demos" info in the "examples-api-use" section) 
For powering I use a power supply from lysonled of 5v 40A (40A are not needed for your setup)
https://bit.ly/2ta8f1l

uros-skopus, 2019-02-20:
Hey man,
Where did you get that small red HUB? Can it be purchased online independent from LED module?

jtgijon, 2019-02-26:
Hi. Yes I bought it at electrodragon. See my comments above
    * An active RPi matrix drive board from electrodragon ( #120 or #406) 
that means:
http://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/
Note that if you use E-line, you must connect the E-line pins accordingly (in the board)

uros-skopus, 2019-02-26:
Yap, I found it meantime. Thanks anyway!
I would appreciate if you could take a look at my latest issue, maybe you can help:
https://github.com/hzeller/rpi-rgb-led-matrix/issues/765

guyverckw, 2019-12-17:
> I forgot that connection. Now it works fine. Thanks,
> [image]
> [image]

Can you show me how you connect the E-line?  I just got exactly the same LED panel as you did, want to test it out, thx!

jtgijon, 2019-12-18:
Hello,
For a 64x64 or 64x128 panel with 1/32 scan, you need to connect Pin 8 to pin E on the Drive Board and P4 to GND.
More information at https://www.hackster.io/idreams/getting-started-with-rgb-matrix-panel-adaa49
Regards

guyverckw, 2019-12-19:
> Hello,
> For a 64x64 or 64x128 panel with 1/32 scan, you need to connect Pin 8 to pin E on the Drive Board and P4 to GND.
> More information at https://www.hackster.io/idreams/getting-started-with-rgb-matrix-panel-adaa49
> Regards

Thanks a lot.

NoicTheBrave, 2021-11-16:
> Hello, For a 64x64 or 64x128 panel with 1/32 scan, you need to connect Pin 8 to pin E on the Drive Board and P4 to GND. More information at https://www.hackster.io/idreams/getting-started-with-rgb-matrix-panel-adaa49 Regards

Hey, all the way in 2021 here:
Is there a way to do this with the Adafruit hat? To be specific: https://www.adafruit.com/product/3211

I was wondering if there was a way for me to do something similar with this hat. Unfortunately, I did not know about the Electrodragon one until it was too late. While I do not mind shelling out a few dollars for a new attachment, if there is a way to resolve this issue with my current hardware, it would be great.

hzeller, 2021-11-16:
As far as I know, Adafruit has not updated their hat. So if you don't want to do the [soldering](https://github.com/hzeller/rpi-rgb-led-matrix#64x64-with-e-line-on-adafruit-hatbonnet), building the [board from this project](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/adapter/README.md) or getting the electrodragon one is probably simpler.

Overboard-code, 2022-04-23:
I have the same 128x64 1/32 panel  I see all of the pictures  Unfortunately the ribbon cable covers the pads on the Electrdragon in all of the pictures.  What did you jumper to get the E line working  8 or 4?

jtgijon, 2022-04-25:
Hello,
I don't have my board here but I remember to connect both. You can check it at https://www.electrodragon.com/w/RPI_RMP_HDK
Regards