hzeller/rpi-rgb-led-matrix #522: 1:4 Outdoor panel multiplex=4 problem

a-bailey, 2018-02-17:
Hello,

I'm trying to get a 1:4 outdoor panel to run with rpi-rgb-led and I am nearly there:

sudo examples-api-use/demo -D3 --led-slowdown-gpio=2 --led-brightness=20 --led-rows=16 --led-multiplexing=4

results in

[image]

I'm a software developer who sadly has no clue about gfx programming and right now I'm trying to make changes to the 4th multiplexer to try to get the result to come out good but nothing I do improved the result. Can somebody give me a hint?

hzeller, 2018-02-17:
Looks like the center stripes of four high need to move outwards and the outer inwards.
So you want to map rows [0..3] -> [5..7], [8..11]->[12..15] and [5..7]->[0..3], [12..15]->[8..11].

a-bailey, 2018-02-17:
I suspected something like that but I'm afraid I can't really get a grasp how I would do that in the transformation. Would I apply this after the multidex? For Example:

use multidex 4 and then afterwards transform the pixel position with a custom transformator? 
Right now I was modifying the sourcecode for the ZStripeTransformer which I think is the wrong approach?

hzeller, 2018-02-17:
Modifying the code of an existing transformer is probabaly the simplest, yes.

angelogoncalve, 2018-02-19:
Try this:

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 4
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'regular'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def matrix_draw(x, y, red, green, blue):
      new_x = x
      if y < 8:
          new_y = y + 8
      else:
          new_y = y - 8
      display.SetPixel(new_x, new_y, red, green, blue)
      return

#while True:
#matrix_draw(1, 1, 0, 255, 0)
#display.SetPixel(1, 1, 0, 255, 0)

for x in range(0, number_of_columns):
for y in range(0, number_of_rows):
matrix_draw(x, y, 0, 255, 0)
#display.SetPixel(x, y, 0, 255, 0)
time.sleep(1)

a-bailey, 2018-02-20:
Thanks for the help, could you show me how the python code especially at the bottom is indented correctly?

angelogoncalve, 2018-02-20:
Use an python editor for indentation for example PyCharm or similar one.

hzeller, 2018-03-17:
Little tip: it is possible to format things correctly in the github issues by formatting it with markdown. So three backticks in the beginning and at the end will help outputting code nicely:
```
  Hello
      World
```

Is this issue solved ?
(check out multiplexing options 4, 5 and 6 which might have been added in the meantime)

closed 2025-11-27 by marcmerlin