hzeller/rpi-rgb-led-matrix #523: P6 32x32 Outdoor panel not displaying properly

rajeshiyengar, 2018-02-18:
Hello,

I am trying to get this 32 x 32 Outdoor Panel working, tried all sorts of combinations with demo image drawer. But none helped so far. Have learnt quite decently using your code and this wonderful forum. Thanks for the code though.

My panel is bit old looks like, it does'nt have the D-Pin. So I have connected it to ground. I verified running various demo. So the pin out looks good.

I am using 6 32x32 Panels. I have connected directly to raspberry pi using your pin out. I have connected 2 parallel. Each parallel connectivity is using 3 32x32 panels. So the chain I am using is 6 in the demos. 

[image]

Below are the images for various demo types.

Running d1 runtext.ppm gave this.
sudo ./demo -D 1 runtext.ppm --led-parallel=2 --led-multiplexing=2 --led-chain=6 --led-cols=32 --led-rows=16 or even with this 
sudo ./demo -D 1 runtext.ppm --led-parallel=2 --led-multiplexing=2 --led-chain=6
[Video Link here](https://youtu.be/Gyist-2jltY)

have compiled the code with the following
--led-slowdown-gpio=2 --led-pwm-lsb-nanoseconds=80 --led-no-hardware-pulse

Output of
sudo ./demo -D 3 --led-rows=16 --led-chain=6 --led-cols=16 --led-parallel=2 
[image]

output of
sudo ./demo -D 3 --led-rows=16 --led-chain=6 --led-parallel=2 --led-multiplexing=0
[image]

output of
sudo ./demo -D 3 --led-rows=16 --led-chain=6 --led-parallel=2 --led-multiplexing=1
[image]

output of
sudo ./demo -D 3 --led-rows=16 --led-chain=6 --led-parallel=2 --led-multiplexing=2
[image]

output of
sudo ./demo -D 3 --led-rows=16 --led-chain=6 --led-parallel=2 --led-multiplexing=3
[image]

output of
sudo ./demo -D 3 --led-rows=16 --led-chain=6 --led-parallel=2 --led-multiplexing=4
[image]

What transformer code will work for me. based on my experiments, it looks like my panel is arranged 16x2 internally. Not sure though.

Please help, wasted a whole week of mine on this.

hzeller, 2018-02-18:
The missing `D` line is expected if this is 1:8 multiplexing, then it only needs to have `A,B,C`. So nothing to worry about here.

In each of the parallel chains, you have 3 panels, so you only need to set `--led-chain=3`. The `--led-multiplexing` option already takes into account the internally half height, so you have to give --led-rows=32 instead of 16. So your baseline configuration is `--led-rows=32 --led-chain=3 --led-parallel=2`

So I suggest to try to see if the any of the following results in a useful output (essentially trying all the led multiplexing
```
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=1
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=2
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=3
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=4
```

Depending on your panels, you might need to add `--led-slowdown-gpio=2` if you have some random pixels show up.

rajeshiyengar, 2018-02-18:
Thanks for the immediate response. 

Meanwhile I tried this #445 and this is the updated pic using his transformer code.
[image]

hzeller, 2018-02-18:
The suggestion with testing `--led-transformer`'s above might be an easier baseline to test, because you then only need relative changes to that (make sure to start with a clean repository without other transformers active).

rajeshiyengar, 2018-02-18:
I tried with all the multiplexing option. the results are same dis-oriented pixels. I have compiled the code with --led-slowdown-gpio=2 and hence not repeating on the demo lines.

the sample output using the fresh repository and running all the below options
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=1
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=2
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=3
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=4

[image]

angelogoncalve, 2018-02-18:
Use one panel only connected and experiment this:

sudo ./demo -D3 --led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=1

rajeshiyengar, 2018-02-18:
Ok, will try that. Meanwhile, completed whitewashed my folder and cloned again and ran the mulitplexing lines. 
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=2
This is what I got

[image]

rajeshiyengar, 2018-02-18:
Output of 
sudo ./demo -D3 --led-rows=32 --led-chain=3 --led-parallel=2 --led-multiplexing=3
[image]

angelogoncalve, 2018-02-18:
Try with only one panel connected to raspberry, it is more easy to see the mapping pixels transform correct that is need for your panel 32x32.

rajeshiyengar, 2018-02-18:
I tried connecting only one panel and it didnt work at all. Moment I connected another 2 panels it is working. Kind of strange, not able to get the logic.

So what do I do now?

angelogoncalve, 2018-02-18:
Use this sudo ./demo -D3 --led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=1

Are you using an hat to control the pixels, use this flag --led-gpio-mapping=adafruit-hat? Or instead use these connections between raspberry and one panel 32x32, use this flag --led-gpio-mapping=regular (see the picture below). 

[image]

rajeshiyengar, 2018-02-18:
When connected with 3 panels, 1 parallel.

sudo ./demo -D3 --led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=0
[image]

sudo ./demo -D3 --led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=1
[image]

sudo ./demo -D3 --led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=2
[image]

sudo ./demo -D3 --led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=3
[image]

sudo ./demo -D3 --led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=4
[image]

rajeshiyengar, 2018-02-18:
With the 
sudo ./demo -D3  --led-chain=1 --led-parallel=1  --led-multiplexing=0 —led-rows=32 --led-cols=16
[image]

rajeshiyengar, 2018-02-18:
I am not using adafruit hat, I am wiring up directly to the RaspberryPi3. I am following hzeller's wiring the pi diagram below
[image]

angelogoncalve, 2018-02-18:
Ok. This option sudo ./demo -D3 --led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=2 was almost good. You need to add this transform I think:

def matrix_draw(x, y, red, green, blue):
    new_x = x
    new_y = y
    if y < 8:
        new_x = x + 32
    elif y < 16:
        new_y = y - 8
    elif y < 24:
        new_x = x + 32
        new_y = y - 8
    else:
        new_y = y - 16
    display.SetPixel(new_x, new_y, red, green, blue)
    return

rajeshiyengar, 2018-02-18:
Thank you so much. 
Where do I add this code? in transformer.cc?

angelogoncalve, 2018-02-18:
You can put that in the function that is responsable for the flag --led-multiplexing=2 using C++ code.

I prefer use the other option using python.

Use this for only one panel connected to the raspberry,  to test a pixel:

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 32
number_of_panels = 1
parallel = 1
number_of_columns = 32

options = RGBMatrixOptions()
options.multiplexing = 2
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows / 1
options.cols = number_of_columns
options.chain_length = 1 * number_of_panels
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'
options.inverse_colors = False
options.led_rgb_sequence = "BGR"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0

display = RGBMatrix(options=options)

def matrix_draw(x, y, red, green, blue):
    new_x = x
    if y < 8:
        new_y = y + 8
    elif y < 16:
        new_y = y - 8
    elif y < 24:
        new_y = y + 8
    else:
        new_y = y - 8
    display.SetPixel(new_x, new_y, red, green, blue)
    return

while True:
    matrix_draw(1, 1, 0, 255, 0)

rajeshiyengar, 2018-02-18:
Ok noted, will try it in python. But can you guide me little in detail as to where I can create this python code and place it. I tried it under bindings/python/samples and nothing came up.

angelogoncalve, 2018-02-18:
Only create a file python with this code, the directory is anywhere you want and use sudo python name_file.py.

Do not forget to install the bindings for recognize the library for python before, I think you have already installed:

cd rpi-rgb-led-matrix
sudo HARDWARE_DESC=regular make install-python

rajeshiyengar, 2018-02-18:
I have done the make install-python and also created the file. There are no errors anywhere. But still python code just hangs and nothing shows up on the panel.

rajeshiyengar, 2018-02-18:
The code provided by you goes into infinite loop as the value of y keeps toggling between 1 and 9 and the terminating condition is never met. Below is how the code looks with proper indentations.
Can you please check and send it again? 

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 32
number_of_panels = 3
parallel = 1
number_of_columns = 32

options = RGBMatrixOptions()
options.multiplexing = 2
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows / 1
options.cols = number_of_columns
options.chain_length = 1 * number_of_panels
options.parallel = parallel
options.hardware_mapping = 'regular'
options.inverse_colors = False
options.led_rgb_sequence = "BGR"
options.gpio_slowdown = 2
options.pwm_lsb_nanoseconds = 80
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0

display = RGBMatrix(options=options)

def matrix_draw(x, y, red, green, blue):
    new_x = x
    if y < 8:
        new_y = y + 8
    elif y < 16:
        new_y = y - 8
    elif y < 24:
        new_y = y + 8
    else:
        new_y = y - 8
    print("X:Y=>{0},{1}",new_x, new_y)
    display.SetPixel(new_x, new_y, red, green, blue)
    return

while True:
    matrix_draw(1, 1, 0, 255, 0)

angelogoncalve, 2018-02-18:
Be attention for the indentations in python. The code for one panel 32x32 chained: 

from rgbmatrix import RGBMatrix, RGBMatrixOptions

number_of_rows = 32
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 2
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'regular'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def matrix_draw(x, y, red, green, blue):
new_x = x
if y < 8:
new_y = y + 8
elif y < 16:
new_y = y - 8
elif y < 24:
new_y = y + 8
else:
new_y = y - 8
display.SetPixel(new_x, new_y, red, green, blue)
return

while True:
matrix_draw(1, 1, 0, 255, 0)

rajeshiyengar, 2018-02-19:
Thanks for the updated code.

Below is the output of the same. Is this was the desired output?
[image]

angelogoncalve, 2018-02-19:
The desired output is fill all the 32x32 pixel a pixel step by step with color green. You need to use only one panel 32x32 at first to see better the mapping pixels of your panel.

Another way is to use this transform in python for only one panel connected to raspberry:

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 32
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 2
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'regular'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def matrix_draw(x, y, red, green, blue):
    new_x = ((x / 16) * 32) + (x % 16)
    if (y / 8) % 2 == 1:
        new_x += 16
    new_y = ((y / 16) * 16) + (y % 8)
    display.SetPixel(new_x, new_y, red, green, blue)
    return

#while True:
#matrix_draw(1, 1, 0, 255, 0)
#display.SetPixel(1, 1, 0, 255, 0)

for x in range(0, number_of_columns):
for y in range(0, number_of_rows):
matrix_draw(x, y, 0, 255, 0)
#display.SetPixel(x, y, 0, 255, 0)
time.sleep(1)

darknessii, 2018-02-20:
Hallo angelogoncalve

It looks like you have the same Panel then me.
you can tray this. with --led-multiplexing=5

https://github.com/darknessii/rpi-rgb-led-matrix

angelogoncalve, 2018-02-20:
Hello Sir. Probably is the same type of panel 32x32 with scan rate 1/8. 

Try to test in your panel with --led-multiplexing=2 with the code below and send the video or pictures please. Thank you for your help my friend. 

If fill all your panel pixel by pixel, your panel is equal as I have. 

Your code support also for python, the flag  --led-multiplexing=5? 

Thank you for your attention my friend.

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 32
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 2
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'regular'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def matrix_draw(x, y, red, green, blue):
new_x = x
if y < 8:
new_y = y + 8
elif y < 16:
new_y = y - 8
elif y < 24:
new_y = y + 8
else:
new_y = y - 8
display.SetPixel(new_x, new_y, red, green, blue)
return

#while True:
#matrix_draw(1, 1, 0, 255, 0)
#display.SetPixel(1, 1, 0, 255, 0)

for x in range(0, number_of_columns):
for y in range(0, number_of_rows):
matrix_draw(x, y, 0, 255, 0)
#display.SetPixel(x, y, 0, 255, 0)
time.sleep(1)

darknessii, 2018-02-22:
Hallo Angelo

you code lookl like this on my Panel, but i hat to do a change on you Python Script.
But if you use the option options.multiplexing = 2, it works not on my changes. 

[image]

options.pwm_bits = 11

[image]

angelogoncalve, 2018-02-22:
Hello Dear friend. 

Thank you for the picture running the code in your panel. 

You say if you use options.multiplexing = 2 and the function below that gives that gif picture in your panel, correct my friend? 

If yes, is another option for mapping correcty this kind of panel 32x32 with scan rate 1/8.

Thanks for your help.

def matrix_draw(x, y, red, green, blue):
new_x = x
if y < 8:
new_y = y + 8
elif y < 16:
new_y = y - 8
elif y < 24:
new_y = y + 8
else:
new_y = y - 8
display.SetPixel(new_x, new_y, red, green, blue)
return

darknessii, 2018-02-23:
Hallo Angelo

Yes, that is what it looks on my Panel, i change the time.sleep(0.02) to speed up the drawing on the Panel.

Yes, it works as a transform for Python.

angelogoncalve, 2018-02-23:
Hello Dear friend. 

Thank you for your atention. 

I see that you have put a new option:

options.multiplexing = 5

That option works in Python? 

Thank you for your help my friend.

angelogoncalve, 2018-02-23:
Hello Dear friend. If you test with your panel sending the gif picture with the option.multiplexing=2 and your code below works also? Thank you for your help my friend

def matrix_draw(x, y, red, green, blue):
new_x = x
if ((y <= 7) or ((y >= 16) and (y <= 23)) or ((y >= 32) and (y <= 39)) or ((y >= 48) and (y <= 55)) or ((y >= 64) and (y <= 71)) or ((y >= 80) and (y <= 87))):
new_y = y + 8
else:
new_y = y - 8
display.SetPixel(new_x, new_y, red, green, blue)
return

hzeller, 2018-03-07:
Can you try if `--led-multiplexing=6` will help work with this panel ?

darknessii, 2018-03-13:
Hallo hzeller

Thank you for writing some Code. I did try, but it is not working on my Panel. May there is some problem with the Mapping. it shows me a lot errors.
Right now, i did not have time to troubleshoot. i hope i find this week some time.

this are my testing results.

[image]

examples-api-use# ./demo --led-rows=32 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=6 -D3
Error in PixelMapper: (0, 8) -> (0, -8) [range: 64x16]
Error in PixelMapper: (1, 8) -> (1, -8) [range: 64x16]
Error in PixelMapper: (2, 8) -> (2, -8) [range: 64x16]
Error in PixelMapper: (3, 8) -> (3, -8) [range: 64x16]
Error in PixelMapper: (4, 8) -> (4, -8) [range: 64x16]
Error in PixelMapper: (5, 8) -> (5, -8) [range: 64x16]
Error in PixelMapper: (6, 8) -> (6, -8) [range: 64x16]
Error in PixelMapper: (7, 8) -> (7, -8) [range: 64x16]
Error in PixelMapper: (8, 8) -> (8, -8) [range: 64x16]
Error in PixelMapper: (9, 8) -> (9, -8) [range: 64x16]
Error in PixelMapper: (10, 8) -> (10, -8) [range: 64x16]
Error in PixelMapper: (11, 8) -> (11, -8) [range: 64x16]
Error in PixelMapper: (12, 8) -> (12, -8) [range: 64x16]
Error in PixelMapper: (13, 8) -> (13, -8) [range: 64x16]
...
...
...
Error in PixelMapper: (24, 31) -> (40, 23) [range: 64x16]
Error in PixelMapper: (25, 31) -> (41, 23) [range: 64x16]
Error in PixelMapper: (26, 31) -> (42, 23) [range: 64x16]
Error in PixelMapper: (27, 31) -> (43, 23) [range: 64x16]
Error in PixelMapper: (28, 31) -> (44, 23) [range: 64x16]
Error in PixelMapper: (29, 31) -> (45, 23) [range: 64x16]
Error in PixelMapper: (30, 31) -> (46, 23) [range: 64x16]
Error in PixelMapper: (31, 31) -> (47, 23) [range: 64x16]
Size: 32x32. Hardware gpio mapping: regular
Press <CTRL-C> to exit and reset LEDs

darknessii, 2018-03-15:
Hallo Henner 

I have the solution for the CoremanMapper. Can you change the CoremanMapper() to this?

class CoremanMapper : public MultiplexMapperBase {
public:
  CoremanMapper() : MultiplexMapperBase("coreman", 2) {}
  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    const bool is_left_check = (x < panel_cols_/2);
    if((y <= 7) || ((y >= 16) && (y <= 23))){
      *matrix_x = ((x / (panel_cols_/2)) * panel_cols_) + (x % (panel_cols_/2));
      if ((y & (panel_rows_/4)) == 0) {
        matrix_x += (panel_cols_/4);
        *matrix_y = (y / (panel_rows_/2)) * (panel_rows_/4) + (y % (panel_rows_/4));
      }
    }else{
      *matrix_x = is_left_check ? x + panel_cols_/2 : x + panel_cols_;
      *matrix_y = (y / (panel_rows_/2)) * (panel_rows_/4) + y % (panel_rows_/4);
    }
  }
};

With 4x 32x32 Panel (2 chain / 2 parallel)
[image]

With 1 Panel
[image]

hzeller, 2018-03-15:
Thanks Roger!
There is this one place where you do
```
matrix_x += (panel_cols_/4);
```

This increments the pointer, which is of course not update the `*matrix_x` value that pointer points to. So if this works as is, then this statement has no effect and could be removed.

Or did you mean to 
```
*matrix_x += (panel_cols_/4);
```
@darknessii ?

darknessii, 2018-03-15:
Hello Henner

Thanks, you got right. This Line can be removed.

Thanks

darknessii, 2018-03-17:
Hello Henner

Thanks for your work. it works fine for me.
great job.

angelogoncalve, 2018-03-19:
Hello Dear Henner.

The --led-multiplexing=6 works fine. Thank you my friend for your help.

rajeshiyengar, 2018-04-17:
@hzeller Thanks a ton, with the --led-multiplexing=6 works absolutely fine. One more help, how can I use 2 different fonts in the runtext.py.

closed 2018-04-17 by rajeshiyengar