hzeller/rpi-rgb-led-matrix #524: 1 master RPI and slave RPI's showing one big picture/video

jsc99dk, 2018-02-18:
Hello

Let's say that I have one raspberry pi controlling 48 LED panels, like this:

<[][][][][][][][]<--------
>[][][][][][][][]              
<[][][][][][][][]<--------  Raspberry PI #1 (slave)
>[][][][][][][][]
<[][][][][][][][]<--------
>[][][][][][][][]

 Then I want to place exactly the setup next to Raspberry PI #1(slave), but controlled by another Raspberry PI #2 (slave).
 And, maybe later place 1 or more setups next to the others setups.
 
 They are then going to show one big video/picture.
 
 I'm thinking of, having one "master" Raspberry pi that are controlling what to display and then have a "slave" Raspberry pi's controlling the LED setups.
 
 Is that possible?
 
 Best regards 
 
 Jesper

hzeller, 2018-02-18:
I'd probably do some sort of networked set-up in which you split the image the way you want it on some master machine and then send the sub-images via the network to Pis. I'd do that with UDP to avoid costly retransmisison times when some packet is lost. Depending on the size of the display, you might need to split the content in multiple 'tiles' as UDP only allows to transmit 64k per datagram.

For these kind of things, I am usually using the [FlaschenTaschen protocol](https://github.com/hzeller/flaschen-taschen/blob/master/doc/protocols.md)  these days. There is also a server implementation that [uses the LED matrix as output](https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display).

jsc99dk, 2018-02-19:
Thank you very much, i will take a look at the FlaschenTaschen protocol.

/Jesper