hzeller/rpi-rgb-led-matrix #525: Flickering Issue with 64x32 Matrix

NYorck, 2018-02-19:


NYorck, 2018-02-19:
I want to connect my pi with an 64x32 Matrix. In between those I've put 7 level shifters (https://www.roboter-bausatz.de/434/i2c-pegelwandler/level-konverter-3.3v-5v-bidirektional-2-kanal) because my Matrix doesnt have the right chip to support the 3,3V from the pi by itself. I've wired everything up at the moment but if I run the runtext example this happens:
https://www.dropbox.com/s/yxeb7cvl7v071wu/20180219_154712.mp4?dl=0
I don't know how to fix that flickering. Could it be that these level shifters don't support the high update frequencies?

maddogs, 2018-02-19:
I did have a similar issue. Try a different power supply and check the grounds are connected.

hzeller, 2018-02-19:
It looks a bit like the values are floating, so make sure to properly connect the signal grounds with the ground on the Pi, level converter and the LED matrix. The 'trails' it pulls look like it is essentially only capacitivly coupled and only works by accident.

The level converter you link to doesn't seem to have a datasheet, so it is hard to give suggestions how to  connect things.

(Given that you have to convert roughly 13 signals, it is probably more economical to use something like 74HCT245)

NYorck, 2018-02-19:
The company that sold the level shifters mailed me this after I asked the for the sheet:
AVCC connect to 5V system power 

ASCL connect to 5V system SCL 

ASDA connect to 5V system SDA 

AGND connect to 5V system GND 

BVCC connect to 3V system power 

BSCL connect to 3V system SCL 

BSDA connect to 3V systems SDA 

BGND connect to 3V system GND 

I connected everything with that information. 
I have to connect all 3 ground pins to ground on the Pi?
And what do you mean it works by accident?

hzeller, 2018-02-19:
'By accident': if the GNDs are floating and not properly connected then you'd see something like that. It would work accidentally because the voltage changes are only transferred via capacitive coupling. If you have connected these all to GND, then this is not your problem.

But if these converters are meant for I2C, I'd check the datasheet for the frequency they can do. Because I2C is happy with only a couple of hundred kHz, but for this LED matrix, you need about 30Mhz for the clocking and with short pulses in the 100Mhz range for the PWM. So your converters might just be too slow if they are only meant to be used for I2C.

So I recommend building [an adapter](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) with standard 74HCT245 or 74AHCT245 chips.

hzeller, 2018-02-19:
Another thing: I2C is meant to essentially be some open-collector circuit, so on one end, things are typically pulled up with a resistor. This might not be sufficient as you now just have a resistor working against the resisudal capacities for the flat cable and the input capacitance of the matrix.
I suspect they only have one transistor or Fet per channel; have a look at the circuit diagram of the converter if they gave it with it.

But for these speeds you really need a push/pull configuration; this is why using the 74-chips are the ones you want to use.

NYorck, 2018-02-19:
I didn't have the grounds from the matrix not connected yet because if I do so this happens:
https://www.dropbox.com/s/5wc4ezrh041ljnk/20180219_174940.mp4?dl=0

NYorck, 2018-02-19:
They didn't provide a circuit diagram

hzeller, 2018-02-19:
Yeah, you need to connect the grounds, don't get discouraged by the worse output, you have to connect all the GNDs to even get to the point where you need to be.

So I suspect the 5V rail you then connect to the Pi or the LED matrix (doesn't really matter), and the 3.3V to the 3.3V output on the Pi (pin 1 on the Pi) ?

Next step is to make sure the power supply on the Panel is stable. If you have a 64x32, make sure your power supply can do at least 6A at 5V. Keep the [power cables short to the matrix]( https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power) (not more than 30cm or so). Depending on the stability of the power supply, you might also want to add a largish (3000uF or so) capacitor right at the power entry of the panel.

Depending on the speed of the panel and/or the voltage converters, you might need to add `--led-slowdown-gpio=2` or higher to the command line options.

NYorck, 2018-02-19:
I connected the matrix with a 5V (The 5V are getting to the matrix, already checked for that) 4A Power supply because the description of the matrix said that it can only take 20W. I take the 3,3V and the 5V Power supply for the level shifters directly from the pi. With all cables grounded and slowdown set to 4 this is the best result I can get:
https://www.dropbox.com/s/uppvem1wjwfep69/20180219_181448.mp4?dl=0

hzeller, 2018-02-19:
So I suspect it is the speed of the level converters and that they can't get the full voltage swing. It looked a bit better when you didn't have the ground, because then the logic level was floating somewhere in the middle of the range. Once you connected the grounds it became clear that the level converters can't drive the 5V swing reliably at the speed.

You could probably play around, reverse engineer the level converters and reduce the resistance values they use for the pull-up. But it is probalby not really worth it as it is unlikely that it will be enough. So I suggest to make your life simple and not use these level converters :)

Just two 74HCT245 or 74AHCT245 [wired like in the adapter (important to wire pin 1 so that the direction is right)](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3) and you're ready to go.

NYorck, 2018-02-19:
the 74HCT245 chips I saw had a direction pin and a an OE pin. Where do I need to connect these to?

hzeller, 2018-02-19:
See the schematics mentioned above.

NYorck, 2018-02-22:
So I ordered the chip soldered everything together and now it looks like this:
https://www.dropbox.com/s/pdyh28s4xx4hl3l/VID-20180222-WA0004.mp4?dl=0
The text has clear edges now but there is a ghosting effect on the left side and there is random colored noise

hzeller, 2018-02-25:
So you clearly have electrical problems, which is hard to debug remotely. 
 
Next steps:

  * If all the signals are connected via HCT245 or AHCT245 (not HC) and operated at 5V, then we don't have to worry about signal integrity anymore (You did add a 100nF capacitor across the power (pin 10 and 20) on each of your both HCT245, right ? Important.)
  * Make sure your power supply is stable and up for the task of supplying of very short pulses of power. [Keep the cables thick and short](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power). Testing a different 5V supply might also give an idea how bad things are.
 * In particular your power supply looks like it can't provide clean power. Add a low ESR capacitor directly at the power inlet of the panel.
[image]
  * Double check that all GNDs are connected.

Das wird schon!

daveglover90, 2018-08-25:
I am also having issues with the 32x64 panel.  I have rewired it about 4 times and continue to have the same issue
Much like the others here is what my issue looks like 
https://www.dropbox.com/s/1v25v4q4ve3l0gc/2018-08-25%2002.02.30.mov?dl=0

Not knowing if the panel was defective I connected it to my Arduino, same wires, same panel power supply.. This is how it looks like with the arduino.

https://www.dropbox.com/s/djn0lvcwekxreqx/2018-08-25%2000.55.22.jpg?dl=0

So I know the panel, wires and power supply are working. 

I have tried a Pi3b and a Pi3b+ with the same issues.  

I have also moved the grounds to a common rail and no changes.

Thoughts

Dave

hzeller, 2018-08-25:
It looks like you are missing a 3.3V -> 5V logic level converter @daveglover90 .
(the Pi GPIO are 3.3V).
https://github.com/hzeller/rpi-rgb-led-matrix#logic-level-voltage-not-sufficient

daveglover90, 2018-08-25:
I tried the led slowdown command, and that got rid of the static.  The text starts to scroll then stops.

Short of building my own, Does the adafruit rgb hat have the 3.3->5 logic level converters in it?

Thanks

Beersatron, 2018-08-25:
I got a completed board from here and it works perfectly

https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

daveglover90, 2018-08-25:
Thanks. I just ordered a couple of them

closed 2025-11-27 by marcmerlin