hzeller/rpi-rgb-led-matrix #527: Single Color P10 Module (Red Color) not working

NeerajShrivastava04, 2018-02-21:
Hi Hzeller,

First of all thank you for writing such a wonderful library.

I am facing an issue while trying to use this library for a single color P10 module (HUB12).
Would highly appreciate you can help me in this regard.

I have gone through the documentation which you have given and have wired my panel in the same fashion.
When I ran a simple pixel test program (given below) on this module then with parameters as rows=8 and chain =1,  LED at Column =1 and  Row = 13, Column =9 and  Row = 13, Column =17 and  Row = 13 and Column =25 and  Row = 13 gets lights up simultaneously. They traverse in the same fashion first horizontally i.e. after these 4 LEDs next sequence comes up with 2,18,14,26 (row 13) and so on.
Once all the 32 LEDs of row 13 gets light up (in 8 steps)  then the same happens for row9 and then for row 5 and then for row 1.

Please find below the video for the same.
First one is with rows=8 and chain=1 
[https://youtu.be/4mqV_CxJLjk](url)
Second one is with rows =8 and chain =2
[https://youtu.be/N3xEg1BNlAg](url)
I have also tried to tweak the different parameters like rows =16 and chain =2 but then only half of my board gets lights in the same fashion.
In case if I put chain =4 then only 1/4th of the panel gets lights up.

I have gone through majority of issues which different users of this library has reported but couldn't figure out the exact mapping for my panel and because of that I am not able to come up with the transformer for this panel.
Also, I have tried the new multiplexing and address type option as well but with no luck.

`// testing 32 x 16 display pixels

#include "led-matrix.h"
#include "transformer.h"

#include <unistd.h>
#include <math.h>
#include <stdio.h>

//using rgb_matrix::GPIO;
//using rgb_matrix::RGBMatrix;
//using rgb_matrix::Canvas;

using namespace rgb_matrix;

static void DrawOnCanvas(Canvas *canvas) {
  /*
   * Really simple animation..
   * Plot pixels one by one. We wait between each step to have a slower animation.
   */

  for (int y = 0; y < 8; y++) {
    for (int x = 0; x < 32; x++) {
                        // Show where we are on screen. Comment out if using fast animation
                        printf("Pixel at (%d, %d)\n", x, y);
      // Draw a pixel at (x, y) where red = 66%, green = 33% and blue = 100% brightness.
      canvas->SetPixel(x, y, 0, 0, 0);
                        // Wait a bit... (increase the sleep time if mapping, decrease for testing).
      usleep(1 * 100 * 1000);  // 1x100x1000 = 100,000 microseconds or 0.1 seconds.
      // Blank this pixel before drawing the next one if mapping, comment out if filling screen
      //canvas->SetPixel(x, y, 255, 100, 100);
    }
  }

}

int main(int argc, char *argv[]) {
  /*
   * Set up GPIO pins. This fails when not running as root.
   */
  GPIO io;
  if (!io.Init())
    return 1;

  /*
   * Set up the RGBMatrix. It implements a 'Canvas' interface.
   */
  int rows = 8;    // May need to change this to 4, 8, 16 or 32
  int chain = 1;    // Number of boards chained together, may need to double for some panels
  int parallel = 1; // Number of chains in parallel (1..3). > 1 for plus or Pi2

  RGBMatrix *matrix = new RGBMatrix(&io, rows, chain, parallel);

  Canvas *canvas = matrix;
  canvas->Fill(255,255,255);

  DrawOnCanvas(canvas);    // Using the canvas.

  // Animation finished. Shut down the RGB matrix.
  canvas->Clear();
  delete canvas;

  return 0;
}

`

NeerajShrivastava04, 2018-02-24:
Further to my investigation on this issue, I am able to light up a single LED at a time by tweaking the parameters of my test program (without transformer).
In the below program I have given  rows = 8 and chain = 4 and by calling setPixel in a loop with x<128 and y<4, I am able to light up the entire 16x32 P10 single color module.
Please see the below video.

https://youtu.be/-8Zv9-vstFQ

Has anyone coded the transformer for this king of panel.
I would highly appreciate any help in this regard.

`// testing 32 x 16 display pixels

#include "led-matrix.h"
#include "transformer.h"

#include <unistd.h>
#include <math.h>
#include <stdio.h>

//using rgb_matrix::GPIO;
//using rgb_matrix::RGBMatrix;
//using rgb_matrix::Canvas;

using namespace rgb_matrix;

static void DrawOnCanvas(Canvas canvas) {
/

Really simple animation..
Plot pixels one by one. We wait between each step to have a slower animation.
*/
for (int y = 0; y < 4; y++) {
for (int x = 0; x < 128; x++) {
// Show where we are on screen. Comment out if using fast animation
printf("Pixel at (%d, %d)\n", x, y);
// Draw a pixel at (x, y)
canvas->SetPixel(x, y, 0, 0, 0);
// Wait a bit... (increase the sleep time if mapping, decrease for testing).
usleep(1 * 100 * 1000); // 1x100x1000 = 100,000 microseconds or 0.1 seconds.

}
}

}

int main(int argc, char argv[]) {
/

Set up GPIO pins. This fails when not running as root.
*/
GPIO io;
if (!io.Init())
return 1;
/*

Set up the RGBMatrix. It implements a 'Canvas' interface.
*/
int rows = 8; // May need to change this to 4, 8, 16 or 32
int chain = 4; // Number of boards chained together, may need to double for some panels
int parallel = 1; // Number of chains in parallel (1..3). > 1 for plus or Pi2
RGBMatrix *matrix = new RGBMatrix(&io, rows, chain, parallel);

Canvas *canvas = matrix;
canvas->Fill(255,255,255);

DrawOnCanvas(canvas); // Using the canvas.

// Animation finished. Shut down the RGB matrix.
canvas->Clear();
delete canvas;

return 0;
}`

NeerajShrivastava04, 2018-06-03:
Hi Hzeller,

I have picked up this issue again after successfully running the RGB matrix with this library.
I am running my HUB12 P10 module with the below parameters : 

`sudo ./scrolling-text-example  -y -2 --led-inverse --led-rows=16 --led-cols=32 --led-chain=2 --led-parallel=1  --led-multiplexing=4 -s 2 -f ../fonts/8x13.bdf ABCD`

With --led-multiplexing = 4 I am getting the attached o/p.

https://youtu.be/LgiGPswO1RM

I have tried different options for modifying the ZStripeMultiplexMapper so as to get it work for this module but with no luck yet.

Would highly appreciate in case if you can guide me through and to make it work.
Please note that I have designed a PCB for this HUB 12 and have reversed the polarity in software.

I know that you have already raised a META BUG [531](url) for this issue but in case if you can help me over here than that would really be great.

Thank you so much once again for such a wonderful library.

NeerajShrivastava04, 2018-08-15:
`Hi Hzeller,

I have written a multiplex mapper for HUB12 panel and have run it successfully (so far only on a single panel. Should work multiple panels as well). Will test the same on multiple panels and will let you know.
Please note that mine is a 32x16 HUB12 single color panel. My panel is internally divided into four parts (128x4). I have written this mapper by keeping my panel in mind and its working fine.
Also, I have enabled the sub-panel flag in lib Makefile.

In case if you feel then you can include this mapper into your library so that others can also use it in case if they have the same panel.

`class P10SingleColorHUB12Mapper : public MultiplexMapperBase {
public:
P10SingleColorHUB12Mapper(const char *name)
: MultiplexMapperBase(name, 4){}

void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
	if (y / 4 == 0) {
		*matrix_y = y % 4;
		if (x / 8 == 0) {
			*matrix_x = x + 24;
		} else if (x / 8 == 1) {
			*matrix_x = x + 48;
		} else if (x / 8 == 2) {
			*matrix_x = x + 72;
		} else if (x / 8 == 3) {
			*matrix_x = x + 96;
		}
	} else if (y / 4 == 1) {
		*matrix_y = y % 4;
		if (x / 8 == 0) {
			*matrix_x = x + 16;
		} else if (x / 8 == 1) {
			*matrix_x = x + 40;
		} else if (x / 8 == 2) {
			*matrix_x = x + 64;
		} else if (x / 8 == 3) {
			*matrix_x = x + 88;
		}

	} else if (y / 4 == 2) {
		*matrix_y = y % 4;
		if (x / 8 == 0) {
			*matrix_x = x + 8;
		} else if (x / 8 == 1) {
			*matrix_x = x + 32;
		} else if (x / 8 == 2) {
			*matrix_x = x + 56;
		} else if (x / 8 == 3) {
			*matrix_x = x + 80;
		}

	} else if (y / 4 == 3) {
		*matrix_y = y % 4;
		if (x / 8 == 0) {
			*matrix_x = x;
		} else if (x / 8 == 1) {
			*matrix_x = x + 24;
		} else if (x / 8 == 2) {
			*matrix_x = x + 48;
		} else if (x / 8 == 3) {
			*matrix_x = x + 72;
		}
	}
}
};

static MuxMapperList *CreateMultiplexMapperList() {
MuxMapperList *result = new MuxMapperList();

// Here, register all multiplex mappers from above.
result->push_back(new StripeMultiplexMapper());
result->push_back(new CheckeredMultiplexMapper());
result->push_back(new SpiralMultiplexMapper());
result->push_back(new ZStripeMultiplexMapper("ZStripe", 0, 8));
result->push_back(new ZStripeMultiplexMapper("ZnMirrorZStripe", 4, 4));
result->push_back(new CoremanMapper());
result->push_back(new P10SingleColorHUB12Mapper("P10SingleColor"));

return result;
}
``

Apart from above changes I have made the following changes as well :

Have changed the Assertion statement in framebuffer.cc:209 to allow 4 rows. Currently its only allowing 8 rows or more than that. Since my panel's internal mapping is 128x4 so I need to make this change as well to allow less than 8 rows.

Have made OE high through software by changing the code in gpio.cc
#define PWM_CTL_POLA1 (0<<4)	// CH1 Polarity (0=(0=low 1=high) 1=(1=low 0=high)

virtual void SendPulse(int time_spec_number) { io_->SetBits(bits_); Timers::sleep_nanos(nano_specs_[time_spec_number]); io_->ClearBits(bits_);

NeerajShrivastava04, 2018-08-15:
Issue got resolved.

closed 2018-08-15 by NeerajShrivastava04

KBNZOKO, 2018-11-04:
Please share with us to help me??

KBNZOKO, 2018-11-05:
Hello! Thank you for what did. Please, I want you to share the modification that you did by "enabling the sub-panel flag in lib Makefile" ?

Thank you for what you did.

hakao32, 2019-02-22:
Hello,
I appreciate what you did, thank you. I want to use your modification for HUB12 but there is a wiring difference between hzeller active-3 adapter and Hub12. hzeller adapter uses HUB75 interface. How you manage that difference?

lbnshrivas, 2019-02-23:
Hi Hakao32....I made my own PCB for testing HUB12 panels.

hakao32, 2019-02-25:
I made another cable to fix HUB12. Thank you any way.
Have you ever try led-pixel-mapper with your P10SingleColor? I need Z-mapper but there is only U-mapper supported with P10SingleColor. How can i do it?
[<][<] i did 64*16 like this. 
But i want to do;
[<][<]
[<][<]
[<][<]
[<][<]
as 64*64. It is Z-map but ZStripeUneven wont work for P10. Any idea?

lbnshrivas, 2019-02-25:
Way back I wrote a mapper which maps 4 rows 

[>]------[>]
[<] -----[<]
[>]------[>]
[<] -----[<]

Let me see in case if I still have that code.

hakao32, 2020-11-12:
This code part is not working for latest version of the library :(