hzeller/rpi-rgb-led-matrix #529: P10 Outdoor Panel 1/4 Scan Problem.

dl2ocb, 2018-02-24:
Hello,

i got some of this nice P10 Outdoor Panels too.

They are 1/4 Scan Panels and i inserted the Code for the P10 Transformation provided by VolkerGoeschl (many many Thanks).

My HW-Config is a Raspberry Pi 3 with the active Adapter for 3 Chains and 1 Panel connected for testing.

I tested the demo Application after inserting the following Line into it:

... Upper Code ....
if (rotation > 0) {
matrix->ApplyStaticTransformer(RotateTransformer(rotation));
}

// INSERTED by DL2OCB --------------------------------------------------
matrix->ApplyStaticTransformer(P10outdoorTransformer());
//------------------------------------------------------------------------------------

printf("Size: %dx%d. Hardware gpio mapping: %s\n",
matrix->width(), matrix->height(), matrix_options.hardware_mapping);
... more Code ....

If i got it right, this should transform the Matrix to be displayed the rigth way on an 1/4 Scan P10 Panel.

I used this commandline:

sudo ./demo -D3 --led-rows=8 --led-chain=4

The Output of the Panel looked like this:

[image]

I dont know what went wrong ... @hzeller can you give me a hint please ?

I also tried the --led-multiplex=4 Option without success.

I can provide you a Panel for testing if you need.

Many thanks for the nice Lib !!

hzeller, 2018-02-24:
It would be easier to insert the P10 transformer into the [place in lib/led-matrix.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/led-matrix.cc#L192) as a new place in the switch/case there, or by replacing one of the switch-case elements that you don't need. Then choose with `--led-multiplex=<thatnumber>`.
Because if you have it in there, you can use the natuarl numbering `--led-rows=16 --led-cols=32` for your panel.

angelogoncalve, 2018-02-24:
Try this and send a video or a picture from the test in your panel 16x32 with scan rate 1/4. Thanks.

sudo ./demo -D3 --led-rows=8 --led-chain=2

int P10outdoorTransformer::TransformCanvas::width() const {
  return delegatee_->width() / 2;
}

int P10outdoorTransformer::TransformCanvas::height() const {
  return delegatee_->height() * 2;
}

void P10outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  int new_x = x;
  int new_y = y;
  int xoffset = 8 * (x / 8);
  int yoffset = ((y + 4) % 8) / 4;
  new_x = x + 8 * yoffset + xoffset;
  new_y = (y % 4) + 4 * (y / 8);
  delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

vapiper, 2018-02-24:
@dl2ocb try to use these parameters `--led-rows=16 --led-cols=32 --led-multiplex=4`

dl2ocb, 2018-02-25:
OK, ill try this Options.
@hzeller  OK, better Idea, ill insert it into the multiplex switch. Thanks for the Idea.
@vapiper I tried this Options before opening this issue.
@angelogoncalve Ill try your Sources too.

Ill report later, busy now ...

CU Stefan

dl2ocb, 2018-02-25:
@hzeller 
Hi,
i inserted the Transformer Code into the location where the other multiplexer Transformers are too (multiplex-transformers-internal.h, multiplex-transformers.cc). I inserted a new --led-multiplexing=5 switch in led-matrix.cc and corrected the options-initialize.cc for the new Option.

I can now select the new Transformer but the Result is not OK.

Here are the Results of the Tests i made:
sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-multiplexing=0
[image]
sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-multiplexing=1
[image]
sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-multiplexing=2
[image]
sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-multiplexing=3
[image]
sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-multiplexing=4
[image]
sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-multiplexing=5
[image]

Looks like the Transformer does not fit the Panel .... any Ideas ?

@hzeller I committed the changes i made.

dl2ocb, 2018-02-25:
@hzeller Here is the diff File of my changes:

[diff_P10_transformer_dl2ocb.txt](https://github.com/hzeller/rpi-rgb-led-matrix/files/1755940/diff_P10_transformer_dl2ocb.txt)

angelogoncalve, 2018-02-25:
Your transformer for your panel 16x32 with scan rate 1/4 is like this:

sudo ./demo -D3 --led-rows=8 --led-chain=2

int P10outdoorTransformer::TransformCanvas::width() const {
return delegatee_->width() / 2;
}

int P10outdoorTransformer::TransformCanvas::height() const {
return delegatee_->height() * 2;
}

void P10outdoorTransformer::TransformCanvas::SetPixel(int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
int new_x = x;
int new_y = y;
int xoffset = 8 * (x / 8);
int yoffset = ((y + 4) % 8) / 4;
new_x = x + 8 * yoffset + xoffset;
new_y = (y % 4) + 4 * (y / 8);
delegatee_->SetPixel(new_x, new_y, red, green, blue);
}

Try this option test and send a picture what happens. Thank you for your attention.
sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=4 --led-row-addr-type=2

dl2ocb, 2018-02-25:
@angelogoncalve I tried the parameters you told me.
The Commandline: sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=4 --led-row-addr-type=2

Output of the Panel:
[image]

schuetzirol, 2018-02-25:
Hey I got also a P10 Outdoor Pannel. I got an 16x96 2 times one underneath the other. I had do make them work for engineer diploma project. I would be very thankful if you can make that work. I have the same problems like @dl2ocb

hzeller, 2018-02-25:
If you are working on your engineering diploma, you probably have access to an oscilloscope to see how things are wired up. So it would be good if you could dig into that and provide a transformer that works for your panel @schuetzirol  Unfortunately, many panels go their own route, so you have to figure that out for your panel ... which might not be the same as others on this thread.
Trying all the `--led-multiplexing` options with the address mode options will help figuring this out. The 96 length is probably a --led-chain=3 --led-cols=32 situation. The under-each-other sounds like you need --led-parallel=2

angelogoncalve, 2018-02-25:
Mapps you panel with a loop and send the video to see what happens my friend. Thank you for your help.

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
    options = RGBMatrixOptions()
    options.multiplexing = 0
    options.row_address_type = 0
    options.brightness = 100
    options.rows = number_of_rows
    options.cols = number_of_columns
    options.chain_length = number_of_panels
    options.parallel = parallel
    options.hardware_mapping = 'adafruit-hat'
    options.inverse_colors = False
    options.led_rgb_sequence = "RGB"
    options.gpio_slowdown = 4
    options.pwm_lsb_nanoseconds = 130
    options.show_refresh_rate = 0
    options.disable_hardware_pulsing = True
    options.scan_mode = 1
    options.pwm_bits = 1
    options.daemon = 0
    options.drop_privileges = 0
    return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

for x in range(0, number_of_columns * number_of_panels):
    for y in range(0, number_of_rows * parallel):
        display.SetPixel(x, y, 0, 255, 0)
        time.sleep(1)

schuetzirol, 2018-02-25:
if i try sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=3 --led-parallel=1 --led-multiplexing=4 --led-row-addr-type=2 
these is what the panel shows:
[image]

angelogoncalve, 2018-02-25:
Dear Schuetzirol try only one panel 16x32 to see what happens and send the pictures please. Tkanks.

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=0 --led-row-addr-type=0

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=4 --led-row-addr-type=2

schuetzirol, 2018-02-25:
sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=0 --led-row-addr-type=0
Output: 
[image]

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=4 --led-row-addr-type=2
Output:
[image]

angelogoncalve, 2018-02-25:
Dear Schuetzirol try this option also only with one panel 16x32 to see what happens and send the picture please. Tkank you for your attention my friend.

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=2 --led-row-addr-type=0

schuetzirol, 2018-02-25:
Thanks for your helping.

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-chain=1 --led-parallel=1 --led-multiplexing=2 --led-row-addr-type=0
Output is:
[image]

schuetzirol, 2018-02-25:
sorry i just saw the pictures from me where upside down.

angelogoncalve, 2018-02-25:
Mapps your one panel 16x32 with a loop and send the video to see what happens my friend. Thank you for your help.

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
display.SetPixel(x, y, 0, 255, 0)
time.sleep(1)

angelogoncalve, 2018-02-25:
Dear Schuetzirol try apply this transformer in one panel 16x32 and send a picture please. Thank you Dear friend.

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
    scan_rate = number_of_rows / 4

    xoffset = (number_of_columns / 4) * (x / (number_of_columns / 4))
    yoffset = ((y + scan_rate) % (number_of_rows / 2)) / scan_rate
    new_x = x + (number_of_columns / 4) * yoffset + xoffset
    new_y = (y % scan_rate) + scan_rate * (y / (number_of_rows / 2))
    display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
transformer(x, y, 0, 255, 0)
time.sleep(1)

schuetzirol, 2018-02-25:
ok thanks I'm going to try it.

schuetzirol, 2018-02-25:
sorry I'm not very familiar with python. So i just copy paste this in any .py file and run it?

angelogoncalve, 2018-02-25:
Ok my driend. Dear Schuetzirol are you using an adafruit-hat or wires connections? If you are no using the adafruit-hat, you need to change options.hardware_mapping = 'regular'

You can also try this another transformer and send also the video test in one panel 16x32. Thanks my friend for your attention.

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
scan_rate = number_of_rows / 4

new_x = ((x / (number_of_columns/4)) * (number_of_columns/2)) + (x % (number_of_columns/4))
if y & scan_rate == 0:
    new_x += number_of_columns/4
new_y = ((y / (number_of_rows/2)) * scan_rate) + (y % scan_rate)
display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
transformer(x, y, 0, 255, 0)
time.sleep(1)

schuetzirol, 2018-02-25:
sorry can you help me by this?
i got this error:     
options = RGBMatrixOptions()
          ^
SyntaxError: expected an indented block

angelogoncalve, 2018-02-25:
Yes Dear Schuetzirol copy for a .py file but be attention for the indentation use an editor PyCharm for windows to help the indentation or the editor that comes in the raspberry (Python2), and run it to test one panel 16x32 connected to the raspberry, and see what happens without an transformer and the two transformers I send to you. 

Send also the three videos to see what happens in the three cases please. 

Thank you for your help my Dear friend.

dl2ocb, 2018-02-25:
@hzeller 
I am using the active RPi adapter with 74HCT parts. I put clean 5V Supply to it and i have 100nF and 10uF extra at Supply-Voltage input.

angelogoncalve, 2018-02-25:
Dear Schuetzirol you need to be attention about he indentation my friend because python do not have {...} to see when begins a block and ends the block. 

def rgbmatrix_options():
(4 spaces)options = RGBMatrixOptions()

schuetzirol, 2018-02-25:
thanks going to solve this. Thanks for your help.

dl2ocb, 2018-02-25:
@angelogoncalve Its easier if you post the Code in a txt File. The formatting will be ok then i think.

hzeller, 2018-02-25:
(Little tip, if you want to exchange code-snippets in bug-entries, you can use markdown formatting, e.g. you can do triple backticks to start a block

```
  int foo = 42;
```
)

dl2ocb, 2018-02-25:
... still no luck ... i fixed the indentation but now i get:

Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import rgbmatrix, rgbmatrixoptions
ImportError: No module named rgbmatrixoptions

The Python Modules are installed by "make install" which starts "python setup.py install".

Any Idea ?

angelogoncalve, 2018-02-25:
Dear Dl2ocb use this:

cd rpi-rgb-led-matrix

sudo HARDWARE_DESC=adafruit-hat make install-python

and this in python

from rgbmatrix import RGBMatrix, RGBMatrixOptions

schuetzirol, 2018-02-25:
solved the python problem. @dl2ocb  you have to run as root, the library need it. I'm still working on filming in a nearly good quality.

angelogoncalve, 2018-02-25:
Dear friend Schuetzirol you are correct. It is need to run as root using sudo. Thanks my friend for your help.

dl2ocb, 2018-02-25:
@angelogoncalve Sorry i got no adafruit-hat so i leave the default.

@schuetzirol can you send me your Script ? I got further errors ... would be nice, thanks.

dl2ocb, 2018-02-25:
@angelogoncalve My errors where:
pi@raspberrypi:~/rpi-rgb-led-matrix/bindings/python/samples $ sudo python test.py 
Traceback (most recent call last):
  File "test.py", line 48, in <module>
    transformer(x, y, 0, 255, 0)
  File "test.py", line 44, in transformer
    display.SetPixel(new_x, new_y, red, green, blue)
UnboundLocalError: local variable 'new_y' referenced before assignment

schuetzirol, 2018-02-25:
`from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
    options = RGBMatrixOptions()
    options.multiplexing = 0
    options.row_address_type = 0
    options.brightness = 100
    options.rows = number_of_rows
    options.cols = number_of_columns
    options.chain_length = number_of_panels
    options.parallel = parallel
    options.hardware_mapping = 'regular'
    options.inverse_colors = False
    options.led_rgb_sequence = "RGB"
    options.gpio_slowdown = 4
    options.pwm_lsb_nanoseconds = 130
    options.show_refresh_rate = 0
    options.disable_hardware_pulsing = True
    options.scan_mode = 1
    options.pwm_bits = 1
    options.daemon = 0
    options.drop_privileges = 0
    return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
    scan_rate = number_of_rows / 4
    new_x = ((x / (number_of_columns/4)) * (number_of_columns/2)) + (x % (number_of_columns/4))
    if y & scan_rate == 0:
        new_x += number_of_columns/4
        new_y = ((y / (number_of_rows/2)) * scan_rate) + (y % scan_rate)
        display.SetPixel(new_x, new_y, red, green, blue)
    

for x in range(0, number_of_columns * number_of_panels):
    for y in range(0, number_of_rows * parallel):
        transformer(x, y, 0, 255, 0)
        time.sleep(1)`

hzeller, 2018-02-25:
(three backticks in a separate line, not only one)

angelogoncalve, 2018-02-25:
Dear Dl2ocb as you not using an adafruit-hat use this: 

options.hardware_mapping = 'regular' 

as send the friend Schuetzirol. 

Thank you for your attention Dear friends.

schuetzirol, 2018-02-25:
```
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
    options = RGBMatrixOptions()
    options.multiplexing = 0
    options.row_address_type = 0
    options.brightness = 100
    options.rows = number_of_rows
    options.cols = number_of_columns
    options.chain_length = number_of_panels
    options.parallel = parallel
    options.hardware_mapping = 'regular'
    options.inverse_colors = False
    options.led_rgb_sequence = "RGB"
    options.gpio_slowdown = 4
    options.pwm_lsb_nanoseconds = 130
    options.show_refresh_rate = 0
    options.disable_hardware_pulsing = True
    options.scan_mode = 1
    options.pwm_bits = 1
    options.daemon = 0
    options.drop_privileges = 0
    return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
    scan_rate = number_of_rows / 4
    new_x = ((x / (number_of_columns/4)) * (number_of_columns/2)) + (x % (number_of_columns/4))
    if y & scan_rate == 0:
        new_x += number_of_columns/4
        new_y = ((y / (number_of_rows/2)) * scan_rate) + (y % scan_rate)
        display.SetPixel(new_x, new_y, red, green, blue)
    

for x in range(0, number_of_columns * number_of_panels):
    for y in range(0, number_of_rows * parallel):
        transformer(x, y, 0, 255, 0)
        time.sleep(1)
```

dl2ocb, 2018-02-25:
Its running now ...

Here is the Video:

https://youtu.be/GhzL1-kFt9E

dl2ocb, 2018-02-25:
@angelogoncalve All OK, i used the "regular" Setting ... this is my Code now:

'''
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():

   options = RGBMatrixOptions()
   options.multiplexing = 0
   options.row_address_type = 0
   options.brightness = 100
   options.rows = number_of_rows
   options.cols = number_of_columns
   options.chain_length = number_of_panels
   options.parallel = parallel
   options.inverse_colors = False
   options.led_rgb_sequence = "RGB"
   options.gpio_slowdown = 4
   options.pwm_lsb_nanoseconds = 130
   options.show_refresh_rate = 0
   options.disable_hardware_pulsing = True
   options.scan_mode = 1
   options.pwm_bits = 1
   options.daemon = 0
   options.drop_privileges = 0
   return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
    scan_rate = number_of_rows / 4
    new_x = ((x / (number_of_columns / 4)) * (number_of_columns / 2)) + (x % (number_of_columns / 4))
    if y & scan_rate == 0:
       new_x += number_of_columns / 4
    new_y = ((y / (number_of_rows / 2)) * scan_rate) + (y % scan_rate)
    display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, number_of_columns * number_of_panels):
    for y in range(0, number_of_rows * parallel):
       transformer(x, y, 0, 255, 0)
time.sleep(1)
'''

dl2ocb, 2018-02-25:
@angelogoncalve All OK, i used the "regular" Setting ... this is my Code now:

```from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():

options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
scan_rate = number_of_rows / 4
new_x = ((x / (number_of_columns / 4)) * (number_of_columns / 2)) + (x % (number_of_columns / 4))
if y & scan_rate == 0:
new_x += number_of_columns / 4
new_y = ((y / (number_of_rows / 2)) * scan_rate) + (y % scan_rate)
display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
transformer(x, y, 0, 255, 0)
time.sleep(1)
´´´

dl2ocb, 2018-02-25:
@angelogoncalve All OK, i used the "regular" Setting ... this is my Code now:

'''
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():

options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
scan_rate = number_of_rows / 4
new_x = ((x / (number_of_columns / 4)) * (number_of_columns / 2)) + (x % (number_of_columns / 4))
if y & scan_rate == 0:
new_x += number_of_columns / 4
new_y = ((y / (number_of_rows / 2)) * scan_rate) + (y % scan_rate)
display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
transformer(x, y, 0, 255, 0)
time.sleep(1)

dl2ocb, 2018-02-25:
@angelogoncalve All OK, i used the "regular" Setting ... this is my Code now:

'''
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():

options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
scan_rate = number_of_rows / 4
new_x = ((x / (number_of_columns / 4)) * (number_of_columns / 2)) + (x % (number_of_columns / 4))
if y & scan_rate == 0:
new_x += number_of_columns / 4
new_y = ((y / (number_of_rows / 2)) * scan_rate) + (y % scan_rate)
display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
transformer(x, y, 0, 255, 0)
time.sleep(1)
```

angelogoncalve, 2018-02-25:
In the transformer function be carefull Dear friends the indentation:

(no spaces - ok)if y & scan_rate == 0:
(4 spaces - change with this number of spaces)new_x += number_of_columns/4
(no spaces - change with zero number of spaces)new_y = ((y / (number_of_rows/2)) * scan_rate) + (y % scan_rate)
(no spaces - change with zero number of spaces)display.SetPixel(new_x, new_y, red, green, blue)

dl2ocb, 2018-02-25:
@angelogoncalve All OK, i used the "regular" Setting ... this is my Code now:
´´´
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():

options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
scan_rate = number_of_rows / 4
new_x = ((x / (number_of_columns / 4)) * (number_of_columns / 2)) + (x % (number_of_columns / 4))
if y & scan_rate == 0:
new_x += number_of_columns / 4
new_y = ((y / (number_of_rows / 2)) * scan_rate) + (y % scan_rate)
display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
transformer(x, y, 0, 255, 0)
time.sleep(1)

angelogoncalve, 2018-02-25:
Send the loop mapping pixels video test with this transformer correctly and try the other I send you also and send also the video test in one panel 16x32 please. 

Thanks Dear friends.

dl2ocb, 2018-02-25:
AHHHH OK 

its running now ... 
Video is here:
https://www.youtube.com/watch?v=r9jOTPEPMOA

dl2ocb, 2018-02-25:
After a some time the Picture is like this:

[image]

dl2ocb, 2018-02-25:
@angelogoncalve OK my Panel is Upside down ...

dl2ocb, 2018-02-25:
I tried agn with shorter delay. Video is here: 
https://www.youtube.com/watch?v=Kupz4f6zFLU

schuetzirol, 2018-02-25:
this code: 
```
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
display.SetPixel(x, y, 0, 255, 0)
time.sleep(1)
```

instant outputs:
[image]

schuetzirol, 2018-02-25:
this gets the same instant output:
```
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows
options.cols = number_of_columns
options.chain_length = number_of_panels
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
scan_rate = number_of_rows / 4

xoffset = (number_of_columns / 4) * (x / (number_of_columns / 4))
yoffset = ((y + scan_rate) % (number_of_rows / 2)) / scan_rate
new_x = x + (number_of_columns / 4) * yoffset + xoffset
new_y = (y % scan_rate) + scan_rate * (y / (number_of_rows / 2))
display.SetPixel(new_x, new_y, red, green, blue)
for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
transformer(x, y, 0, 255, 0)
time.sleep(1)
```

dl2ocb, 2018-02-25:
With this Code i got this:
https://www.youtube.com/watch?v=Tz9S7Wzy13w

I changed this:
   options.multiplexing =2 
    options.row_address_type =2 

```from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16 
number_of_panels = 1
parallel =1 
number_of_columns = 32

def rgbmatrix_options():
    options = RGBMatrixOptions()
    options.multiplexing =2 
    options.row_address_type =2 
    options.brightness = 50 
    options.rows = number_of_rows
    options.cols = number_of_columns
    options.chain_length = number_of_panels
    options.parallel = parallel
    options.hardware_mapping = 'regular'
    options.inverse_colors = False
    options.led_rgb_sequence = "RGB"
    options.gpio_slowdown = 4
    options.pwm_lsb_nanoseconds = 130
    options.show_refresh_rate = 0
    options.disable_hardware_pulsing = True
    options.scan_mode = 1
    options.pwm_bits = 11
    options.daemon = 0
    options.drop_privileges = 0
    return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
    scan_rate = number_of_rows / 4
    new_x = ((x / (number_of_columns/4)) * (number_of_columns/2)) + (x % (number_of_columns/4))
    if y & scan_rate == 0:
        new_x += number_of_columns/4
        new_y = ((y / (number_of_rows/2)) * scan_rate) + (y % scan_rate)
        display.SetPixel(new_x, new_y, red, green, blue)
    

for x in range(0, number_of_columns * number_of_panels):
    for y in range(0, number_of_rows * parallel):
        transformer(x, y, 0, 255, 0)
        time.sleep(0.1)

schuetzirol, 2018-02-25:
used your code and got a slightly different output:
https://youtu.be/hUMNfH5TPKk

angelogoncalve, 2018-02-25:
I forgot to tell to try this options my friends in rows an chain_length to work.

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows/2
options.cols = number_of_columns
options.chain_length = 2*number_of_panels
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
scan_rate = number_of_rows / 4

new_x = ((x / (number_of_columns/4)) * (number_of_columns/2)) + (x % (number_of_columns/4))
if y & scan_rate == 0:
(4 spaces)new_x += number_of_columns/4
new_y = ((y / (number_of_rows/2)) * scan_rate) + (y % scan_rate)
display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
transformer(x, y, 0, 255, 0)
time.sleep(1)

angelogoncalve, 2018-02-25:
I forgot to tell to try this options my friends with another transformer in rows an chain_length to work also.

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
options = RGBMatrixOptions()
options.multiplexing = 0
options.row_address_type = 0
options.brightness = 100
options.rows = number_of_rows/2
options.cols = number_of_columns
options.chain_length = 2*number_of_panels
options.parallel = parallel
options.hardware_mapping = 'adafruit-hat'
options.inverse_colors = False
options.led_rgb_sequence = "RGB"
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.show_refresh_rate = 0
options.disable_hardware_pulsing = True
options.scan_mode = 1
options.pwm_bits = 1
options.daemon = 0
options.drop_privileges = 0
return options

options = rgbmatrix_options()
display = RGBMatrix(options=options)

def transformer(x, y, red, green, blue):
scan_rate = number_of_rows / 4

xoffset = (number_of_columns / 4) * (x / (number_of_columns / 4))
yoffset = ((y + scan_rate) % (number_of_rows / 2)) / scan_rate
new_x = x + (number_of_columns / 4) * yoffset + xoffset
new_y = (y % scan_rate) + scan_rate * (y / (number_of_rows / 2))
display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, number_of_columns * number_of_panels):
for y in range(0, number_of_rows * parallel):
transformer(x, y, 0, 255, 0)
time.sleep(1)

angelogoncalve, 2018-02-25:
Use only one panel 16x32 Dear friend Schuetzirol connected to raspberry to test the last two codes options I send. Thank you my friend and send the two videos to see if it works changing the rows and chain_length options.

schuetzirol, 2018-02-25:
The last two codes leaf the Panel Black.

angelogoncalve, 2018-02-25:
Dear friend Schuetzirol are you using only one panel 16x32?

[46 more replies]

closed 2018-03-17 by hzeller