hzeller/rpi-rgb-led-matrix #530: shifted pixels

Toootless, 2018-02-25:


hzeller, 2018-02-25:
this is indeed interesting, as things of the upper half of the display suddenly show up in the lower half. They are typically strictly separated (after all, one part gets data from R1, G1, B1, the other from R2, G2, B2).

So I have the suspicion that something went wrong with the panel; less likely is some weird cross-talk between GPIO pins going on (is this hand-wired or through a logic level converter ?). I have seen GPIO pins going wonky when accidentally shorting some pins to something, then all bets are loose (Making sure to use level converters is always a good idea, just to shield the delicate GPIO pins.)

In general make sure that nothing else software wise on the PI uses GPIO pins in any way, e.g. 1-wire communication or i2c (the typical things recommended in the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting).

Toootless, 2018-02-25:
I changed the font size and color
[image]

hzeller, 2018-02-25:
(in order to get a little more cleaner pixels, I suggest to play with increasing the `--led-slowdown-gpio` value and maybe try `--led-pwm-lsb-nanoseconds=200` or something. But that are just independent remarks, they are probably not affecting your initial problem).