hzeller/rpi-rgb-led-matrix #532: raspberry pi VS C10 mainboard

beikimajid, 2018-02-26:
Hi
I compared raspberry with the C10 motherboard led matrix and I saw a movie with each one. The intensity of the movie light is low in the raspberry(80% less than C10) and its flicker is higher with the unobtrusive eye (mobile camera).
Is there a way to change overclock(as example add arm_freq=1200) and get better The intensity of the movie light.

russdx, 2018-03-17:
The HD-C10 will be using a dedicated real time FPGA or powerful micro controller which can continually refresh the display very fast without being interrupted. The raspberry pi is using a cpu which has other tasks to do as well which takes time away from refreshing the display so the result will not be as good  (but very close)

Regards
Russell

hzeller, 2018-03-17:
What Russell said: A dedicated card will always be better than this project. In this project, we just leverage what we can get from the Pi for free. So yes, if you want the highest performance, optimized for the particular kind of panels you are using: use a dedicated card.

An RT kernel might help, but it depends if it is not a limit in memory bandwidth in the processor that results in flicker. If it is an hardware issue, the RT kernel will not help.

Some people tried overclocking, but found that it might make things not neccesarily better #41 .

The lower brightness is actually interesting: it comes from some limitation in how fast a particular timing circuit I am using in the Pi can be re-initialized only with a short down-time (though if someone knows how to work-around that, that would be great).

In general, if you want bright displays _and_ high refresh rate I strongly suggest to use one of these 'outdoor panels'. They have a 1:8 multiplexing (32x32) or 1:4 multiplexing (32x16) and are generally allowing for at least double the refresh rate than regular panels. And they are insanely bright. Displays with 32x32 and 1:16 multiplexing or even 64x64 with 1:32 multiplexing can never reach that.

There are some timings tricks that can be implemented for a particular type of panel that can make the refresh performance of a Pi faster, but the code is done so that it should work with all kinds of panels, so it necessarily has to be conservative. But if you have measured the maximum performance of the panel you're using and know how fast pulses it can do, you can optimize for that (e.g. `--led-slowdown-gpio=0` is too fast for many panels).

Anyway, closing this now, I guess nobody expects that a general purpose cheap hardware can bring the same performance as a dedicated hardware solution.

closed 2018-03-17 by hzeller

beikimajid, 2018-03-25:
Can you explain the meaning of a dedicated card?
Is it connected to the raspberry?
Is it possible to use the raspberry as the sender and receiver in the market as receiver(as example HD-R500 Or Linsn RV908 ?
[image]

hzeller, 2018-03-25:
Dedicated cards come in all kinds of forms. They have pixel sending and timing implemented in hardware using some FPGAs and have all kinds of inputs from HDMI directly to some proprietary protocols that only run on some windows machines.