hzeller/rpi-rgb-led-matrix #534: A outdoor module with really strange pixel mapping (modules from Absen C9 led screen panels)

depili, 2018-02-26:
This is an educational tale of some strange panels and getting them to work. Since their availability is pretty non-existential this will not probably be useful to others as-is, just thought you might be interested to hear about this.

Helsinki Hacklab recently came into possession of dozens of modules from Absen C9 led screen system ( http://www.usabsen.com/mobile-creative-led/ ). The panels are 8x64 1/4 scan 9mm pitch. We are still wondering what to use them for, but this is the tale of their properties.

Internally they are made of two pcbs, one is the left half and one the right. Each pcb has a 2x8 pin 2.54mm pitch male row header for input ( https://www.flickr.com/photos/helsinkihacklab/40242725421 ). It takes one set of rgb, a,b (and not connected c), clk, latch and oe signals + 3 pins of vcc and 4 pins of gnd.

So one such module can be driven from one hub75 input signal routing the rgb1 to one side and rgb to the other. I chose rgb1 for the right side viewed from the back of the modules.

The pixel mapping for the modules is the weirdest I have heard of. First of all the mapping can be broken into 4x4 squares of leds for a half panel. The squares snake through the panel halves like this:

| | | | |
| ---      | ---       | ---       | ---       |
| 0 | 3 | 4 | 7 |
| 1 | 2 | 5 | 6 |

Inside each block the x coordinate is inverted, meaning that the actual top left pixel is x=3, y=0 from the hub75 data.

And because the way the halves are made as different pcbs rows 0-3 will go to the left half and 4-7 to the right.

The one major drawback of the modules is that they have no data output at all and thus can't be chained. So to make controlling them little bit easier we decided to make two such panels behave like 1/8 scan 16x64 module by adding little logic to the c and oe lines (using one 7402 quad 2-input nor gate ic) that we get a truth table like this

|OE |C |OE1 |OE2 |
| ---      | ---       | ---       | ---       |
| 0 | 0 | 0 | 1 |
| 0 | 1 | 1 | 0 |
| 1 | 0 | 1 | 1 |
| 1 | 1 | 1 | 1 |

Using this arrangement the mapping again changes a little with rows 0-3 going to the top-left 8x32, 4-7 to bottom-left 8x32, 8-11 to top-right and 12-15 to bottom left.

This whole stack of weirdness can finally be addressed by the kludges in https://gist.github.com/depili/86e79755df38fb38982c63de4300c90f

depili, 2018-03-09:
And I now have the first PoC 6 module unit running: https://www.flickr.com/photos/helsinkihacklab/26837490508/

The first revision of the adapter boards are quite usable, but will do minor tweaks on the final 48 board batch: https://www.flickr.com/photos/helsinkihacklab/38897960460/

hzeller, 2018-03-09:
Nice work!

Sometimes figuring out the mappings and inner workings of LED panels of unknown origin and without documentation is the most frustrating, but when done, most satisfying thing.

To build pixel mappers for single panels should be relatively easy these days by adding them to the [multiplex mappers](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc). That way, you then have the mapping happening underneath and use the canvas just as a nice square thing.

depili, 2018-03-10:
I wrote an older style static transformer, which is basically just one 16x64x2 look up table, as we might at some point use fpga's to drive the modules and such a table is the easiest implementation there.

depili, 2018-03-10:
The reason I didn't take a look into the multiplex transformers was that trying to test them with a single 8x64 panel just failed assertions about the panel size, but otherwise the multiplex transformer hooks seem to be easier than the old static transformers.

depili, 2018-05-10:
![1/4rd of the final display]([url=https://flic.kr/p/25VBt7L](https://farm1.staticflickr.com/911/41300814044_f0610a8624_k.jpg)

A test of 1/4 of the final display. Our coder wanted to do some hdr tricks so the panels get 16bits per color per pixel, 11bits timing and some dithering on top of that for the whole 16bit linear colorspace... This results in some noise in the picture but improves the dark colors significantly.