hzeller/rpi-rgb-led-matrix #537: Python: PermissionError writing files to drive after calling 'RGBMatrix'

cultpenguin, 2018-03-02:
Hi. 
First, thank you for a very nice piece of software. 

I am using the python interface, to plot the results of some simulation over time.
 Basically I have a loop, where in each loop i run a small program (that needs to write a file to the harddrive) and then it should be visualized using matrix.SetImage. 

However I find that after I have called 'RGBMatrix' I can no longer write to the drive, I get permission error :"PermissionError: [Errno 13] Permission denied: 'file2.npy'", using e.g.:
'''
np.save('file1',myarray)
matrix = RGBMatrix(options = options)
np.save('file2',myarray)
'''

Is there anything I can do such that I can still have write access to files after 'RGBMatrix' has been called?

- Thomas

hzeller, 2018-03-17:
The RGB matrix initializes the hardware and then switches to a different, lower privileged user, so that you don't run as root. But that also means, that files you write will be written as that user.
So either set `options.drop_privileges = False` to continue running as root, or don't write files before the privileges are dropped :)

OllmGrb, 2020-03-06:
Re the above.
I connected the sda and scl to the respective pins, which shouldnt be used by the panels as far as I can tell. I also set options.drop_privileges = False but to no avail. At which point are the privileges dropped, currently working in graphics.py. cheers

closed 2025-11-27 by marcmerlin