hzeller/rpi-rgb-led-matrix #539: With Pizero W

hoangphan, 2018-03-04:
hi @hzeller ,

first of all, i'm very grateful to see your contribution of such a great library.
i'm planning to use your library into some IOT project that can control RGB Panel wirelessly.

however, i'm stuck to make the led on and run your demo :) although i have studied your documentation very well.

can you please hint if this library works with PI ZERO W version 1.3? 

for your information about my bench,
- Raspbian stretch lite 29-Nov-17
- Pi zero W version 1.3
- power supply 5V, current adjustable up to 30A
- PI connected with my MacbookPro via USB, SSH command line only, no GUI
- wiring from PI to LED directly as per the wiring table you shared. i didn't make any level shifter.
- RGB panel 64x64, with E pin in the pin 8. scan rate 1:32

any help would be highly appreciated :)
Hoang

hoangphan, 2018-03-05:
Or did i miss to pull up the OE by 1K resistor?

hzeller, 2018-03-05:
Pi zero w should work, but haven't tried it (sometimes Raspian messes up GPIO detection, but it _should_ just be recognized as Pi1 internally)
Pull-up not really needed, only helps to not light up LEDs when GPIO is floating on reboot.

Usually, I would suspect the missing level shifter.

hoangphan, 2018-03-07:
Thanks for your response! 
i ordered the PCB and would have it by next Tuesday. Hopefully i have some good news for the PI zero W :)

hoangphan, 2018-03-07:
@hzeller : In the meantime, i just took a close look to panel and see that i have 64x64 Led panel P3, with ICN2038 led drivers, and 74HC245. HUB75 has E line pin 8 as per what described as supported from your library.

with the 245, i tried to slow down GPIO and use 4V input (even lower), but without success.

with the ICN2038s, is there any one here success with it?
[image]
[image]
[image]

hzeller, 2018-03-07:
I think there is some discussion over there on #466
It might need some smallish patch, but it is not quite clear yet if it will be the final thing.

hoangphan, 2018-03-09:
@hzeller: i think this topic can be closed. Thanks very much for your support :)

a suggestion in the readme of your project: is to mention about the supported led drivers chip support (i think a poll around would help to consolidate this real quick :)

closed 2018-03-09 by hoangphan

andrei0807, 2019-08-22:
@hoangphan 
How are you?
Are you fine with icn2053 chipset?
How did you patch library for icn2053?

daveythacher, 2020-09-27:
These will require a restructure of the code base. You will need to update the objects to support extending them. Then you would need to add logic for the new driver model. This could break the existing interface.

Updater and DumpToMatrix will need to be overhauled. Currently these are kind of split between two objects. It is possible to make this work, but there is a few issues that would need to be considered. 

I would not recommend it, the restructuring alone makes a new code base easier. The interface for this project is built around strange BCM notions implemented in user space bit bang.

daveythacher, 2020-09-27:
Where did you get one of these panels by the way?