hzeller/rpi-rgb-led-matrix #544: how to create own Image ppm file in rgb

ManjunathRN, 2018-03-06:
Hello sir,
I am new to git hub
I'm working on HUB75 for a project with Pi, and i am using your client server code with 1:16s scan rate , HUB75 board each four 32x32 panel connected in series and parallel .I have did different types of text scrolling and yours ppm files as included in this flaschen -taschen but i don't know how to create own ppm image file and run it in rgb .please help me actually i am new one to this
Thank you

hzeller, 2018-03-06:
You don't have to do it manually, there is extensive API support.

There is an [C++ API](https://github.com/hzeller/flaschen-taschen/blob/master/api/include/flaschen-taschen.h) (that includes features such as simple graphics and text) and a [Python API](https://github.com/hzeller/flaschen-taschen/blob/master/api/python/flaschen.py) that allow to do that programmatically. There are various [examples](https://github.com/hzeller/flaschen-taschen/tree/master/examples-api-use) that help you get started.

The ppm image format is very simple if you want to create it yourself, you find the description in the [protocols](https://github.com/hzeller/flaschen-taschen/blob/master/doc/protocols.md) description.

hzeller, 2018-03-10:
(also, please file a bug in the FlaschenTaschen project if you have problems there not here).

closed 2018-03-10 by hzeller