hzeller/rpi-rgb-led-matrix #546: [64x32 panel] 1st and 3rd quarter not showing

ct4711, 2018-03-06:
Hello!

**First at all thanks for this great project!**

I have a 64x32 panel from aliexpress (with some really bad soldered resitors) where the 1st and 3rd quarter of the display not showing anything.
The connections are double-checked and i think everything is connected correct.
I have also tried many combinations of the led-multiplexing and led-row-addr-type without success.

Do i have a defective or unsupported panel?

Thanks in advance
ct4711

[image]
[image]
[image]

hzeller, 2018-03-10:
If you have everything connected, then this looks indeed like a defective panel unless they have some different multiplexing. You do have a different color mapping for some reason, but you can change that with `--led-rgb-sequence`.

Do you have a close-up picture of input connector and any markings it might have ?

ct4711, 2018-03-12:
You are absolutly right, the right color mapping for this panel is RBG.

The connector doesn't have any markings, but now i'm 100% sure that the panel is defective:
If i bend the pcb a bit the 1st and 3rd quarter are flickering.

Thank you for your response and clearing out that it has to be a hardware problem.

closed 2018-03-12 by ct4711