hzeller/rpi-rgb-led-matrix #548: Output Display blinks instead of having single manual physical gpio button for on & off

henrylowzh, 2018-03-07:
Hi there,

I have created a manual trigger button(on/off) gpio button for my output. I took the "image-viewer.py" sample code as a reference. For your infomation I'm using a 64x128 display with a E line and had perform the pwm mod on the adafruit-hat.

When i launched the script, the gpio button automatics triggers both On & Off(getting a undesired Blink Display) without me pressing anything. I'm curious is there anything on the rgbmatrix module conflicting with the gpio? Or is there somewhere wrong in the code?

please find my code below:

```
import time     # Import sleep Library
import sys
import RPi.GPIO as GPIO    # Import GPIO Library
from rgbmatrix import RGBMatrix, RGBMatrixOptions
from PIL import Image

state = True
size = 64, 64

GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)   # Use Physical Pin Numbering Scheme
button1=21                 # Button 1 is connected to physical pin 16

GPIO.setup(button1,GPIO.IN,pull_up_down=GPIO.PUD_UP) # Make button1 an input, Activate Pull UP Resistor

BS1=False                  # Set Flag BS1 to indicate LED is initially off

image_file1 = '/home/pi/rpi-rgb-led-matrix/danger.jpg'
image1 = Image.open(image_file1)
imageprop1 = image1.convert("RGB").rotate(180)

options = RGBMatrixOptions()
options.rows = 64
options.cols =128
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'adafruit-hat-pwm'  # If you have an Adafruit HAT: 'adafruit-hat'

matrix = RGBMatrix(options = options)

while(1):                  # Create an infinite Loop
        if GPIO.input(button1)==0:            # Look for button 1 press
                if BS1==False:                # If the Display is off
                        print"Button 1 was Pressed to Turn Display ON"
                        BS1=True              # Set Flag to show Display is now On
                        image1.thumbnail(size, Image.ANTIALIAS)
			image1.load()
			matrix.SetImage(imageprop1,32,0,)
			time.sleep(.5)             # Delay
                else:                         # If the Display is ON
                        print"Button 1 Was Pressed to Turn Off"
                        matrix.Clear()
                        BS1=False               # Set Flag to show Display is now Off
                        time.sleep(.5)

```

henrylowzh, 2018-03-10:
Hi @hzeller,

Thank You for your advice. For starters edited the GPIO setup after the matrix initialization
```
matrix = RGBMatrix(options = options)
GPIO.setup(button1,GPIO.IN,pull_up_down=GPIO.PUD_UP) 
```
However, i get a runtime error whereby
```
File "worktest4.py", line 27, in <module>
    GPIO.setup(button1,GPIO.IN,pull_up_down=GPIO.PUD_UP) # Make button1 an input, Activate Pull UP Resistor
RuntimeError: No access to /dev/mem.  Try running as root!
```
even though i had run the script on sudo(root)

I had also checked on the permission access for gpiomem
```
crw-rw---- 1 root gpio 244, 0 Mar  8 11:41 /dev/gpiomem
```
I might have a mistake somewhere. Could you advice?

hzeller, 2018-03-10:
The default led matrix start drops the privileges. Can you set the drop_privileges value in the runtime options to false ?

henrylowzh, 2018-03-10:
Hi @hzeller 

That totally works. The button is working as expected.
However the display image on the led matrix turn out wasn't good. The display color seem to be stretch and there were some slight flickering. Has this anything got to do with the refresh rate or pwm?
[image]

Compare to the earlier display which was good but was blinking
[image]

hzeller, 2018-03-10:
mmh, I don't know what the GPIO library in Python does to the rest of the system, hopefully it just reads the value, but it _looks_ like it messes with the data.
I can't really help here, as I usually do all my GPIO stuff in C++, but maybe someone else knows if there are weird interactions with the GPIO library in Python ?

henrylowzh, 2018-03-10:
Alright Sure~
Thank You very much for your help though~

hzeller, 2018-03-10:
Maybe the very fast polling in your endless-loop is creating too much memory bandwidth pressure on the GPIO. Try to sleep for 100ms between your loops. If this is for interactive input from users, this will be fast enough, but it will not mess with the GPIO subsystem tens of thousands of times a second.

angelogoncalve, 2018-03-10:
Try this:

options = RGBMatrixOptions()
options.rows = 64
options.cols =128
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'adafruit-hat-pwm'

options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.pwm_bits = 1

henrylowzh, 2018-03-11:
Hi @angelogoncalve Hi @hzeller 

I took your advice changed the delay to at least 100ms and added in the 
```
options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds = 130
options.pwm_bits = 1
```
the modifications kinda improve the display quality. However, looks like i'm forced to change the 
```
options.hardware_mapping = 'adafruit-hat-pwm'
```
 to 'adafruit-hat' as with -pwm i'm getting distorted displays.

Thank you very much guys. Appreciate it.

angelogoncalve, 2018-03-11:
Yes you are correct I forgot to tell you about this option to change also. Sorry my friend.

options.hardware_mapping = 'adafruit-hat'

hzeller, 2018-08-02:
So looking at your first code again: you are using GPIO21 as input, but that is actually being used by the Adafruit HAT for strobe. So whatever you read there might indeed be wonky. Only GPIO inputs 2 3 7 8 9 10 11 14 15 19 24 25 are available.

closed 2018-08-02 by hzeller