hzeller/rpi-rgb-led-matrix #549: scan mode and refresh rate

Ellhassan213, 2018-03-07:
Hi,
Is it possible with this library to drive the Matrix statically instead of using the default scan mode?
I cannot achieve the refresh rate as high as i want, so i would like to get rid of that problem.

I am observing the LED lights through hammatsu orca scientific camera, unfortunately the pictures i am getting captures the frequency at which the LEDs are refreshed.

So is there a way to avoid refreshing the matrix?

Thanks very much

hzeller, 2018-03-07:
Hey, don't ask the same question on different issues, that just uses time for anyone involved in answering these (like here: #49 ).

To repeat what I said there:
The panels need update all he time as they don't have any framebuffer; the Pi has to even do the PWM for each pixel. Even if you only do one color mode without PWM, the panels need to be scanned through all available lines one-by-one, as only one line can be lit at a time. So no, these kind of panels do not have a way to just statically display a picture.

Your best chance at a high update rate is to get some outdoor panel with 1:8 scanrate (32x32) or even better 1:4 scanrate (32x16 panels).Don't use 64x64 panels, they are slow.
Also, if you don't need too much color resolution, lowering the `--led-pwm-bits` can slightly increase the refresh rate.

As usual with watching LED screens with a camera, you need to adjust your exposure time to be longer than the refresh rate (you can see the refresh rate with `--led-show-refresh`). Typically, I'd make it double or triple; so if you have a refresh rate of 900Hz (typical with one of these 1:4 panels), then make the exposure not shorter than 1/300s.
Often, cameras are overwhelmed with the brightness of the LEDs and react with shortening the exposure time; you can prevent that with stopping down the aperture or with photographic gray filters.

Ellhassan213, 2018-03-07:
Hi thanks very much, I will look into what you suggested.

I will let you know of any progress in the next few days.

Sorry about positing twice, was a mistake, i thought the first one did not work as it disappeared off my screen somehow.