hzeller/rpi-rgb-led-matrix #550: Adafruit RGB Matrix HAT + RTC for Raspberry Pi

DebaGracias, 2018-03-08:
Hi All,
I purchased a Adafruit RGB Matrix HAT + RTC and mounted it on my Ras Pi 3 .I followed all the steps mentioned in the [link ](https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi).However when I try to run my code using this library ,I dont see the desired result.I just see few lines of leds lit.I dont know what i'm doing wrong.
I ordered the LED board from [AliExpress](https://www.aliexpress.com/item/40-inch-101cm-ios-And-Android-Wifi-wireless-remote-Programmable-Advertising-LED-Display-Board-Bright-Red/32752577115.html?spm=a2g0s.9042311.0.0.F3UzUv)
I have also attached an image of Input Pin diagram of LED board and an image showing the issue.
[image]
[image]

Could it be that the HAT is not compatible with this LED board?

hzeller, 2018-03-08:
indeed, looks like you got a red/green panel with hub12 connector, not RGB with hub75. You might need to do some manual wiring and inversion of OE to get it to work with this library.

hzeller, 2018-03-10:
We can probably combine this with the collecting bug for Hub12 requests #531 and close this one to keep things simple.

closed 2018-03-10 by hzeller