hzeller/rpi-rgb-led-matrix #551: Overclocking and Flickering (Reply)

beikimajid, 2018-03-08:
Hi
according to : [https://github.com/hzeller/rpi-rgb-led-matrix/issues/41](url)
I test raspbian with last RT_Kernel but see no diffrence.I run led-matrix it will flicker (p5 smd & 3 Rows & 6 Cols With passive PCB).

I even test these conditions:

remove overclock setting
add overclock setting
remove isolcpus=3 in RT-Kernel
addisolcpus=3 in RT-Kernel
Are you sure that RT_Kernel is answer to have not flicker?

hzeller, 2018-03-08:
RT kernel can improve flickering, but of course not eliminate it if we hit a hardware limit. So far the question is still open if it can create a much better experience than the regular kernel, so these experiments are needed.

Never do overclocking and always use isolcpus independent of the kernel.

You did not describe details to help people reproduce your findings: what compile-time parameters you choose when you compiled the kernel, or if it is a stock kernel, which exact distribution ? What kernel parameters did you choose at boot time (such as NO_HZ) ?

 What did you do to induce the flicker? Usually flicker can only be seen these days if you run something in another shell.

russdx, 2018-03-16:
I will be trying a RT kernel soon to try and get 100% flicker free experience.

Is it as simple as just running RPI-RGB on a RT kernel with isolcpus=3 or does the RPI-RGB process need to be setup in a way it takes advantage of the RT kernel? ie it says it needs to be real time and do not every interrupt it.

Regards
Russell

hzeller, 2018-03-17:
Check out the new compile-time option in `FIXED_FRAME_MICROSECONDS` in [lib/Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105). It will allow to time-buffer glitches created by the rest of the system.

HummingBuddha, 2018-07-04:
Hello hzeller
I am using a 64x64 32s P2.5 module. I have too much of flicker. I tried "FIXED_FRAME_MICROSECONDS in lib/Makefile" but of no use. Is there any ways to improve the scan rate programatically? In other scan issues you have mentioned not to use 64x64 32S modules as they are slow but I have no other go I am working on a 64x64 P2.5, so pls let me know is there any solution for the flickers. 
And is there any ways to compress and accommodate larger image files into the UDP size?

Thanks

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828