hzeller/rpi-rgb-led-matrix #552: 5 panels 32x16

ostium, 2018-03-08:
Hello,
I'm trying to build a video-wall with 5 panels 16x32, like image below, with 480mm (length) x 192mm (height).

[image]

It is possible to do this? 

I have all panels wired up, and i can execute "sudo examples-api-use/demo -D0" without problems, but now I don't know how to configure this setup.

Please, can someone help me?

Thanks in advance :-)

hzeller, 2018-03-08:
Yes, you can do this. I suppose you have 5 panels with 32x16 and you chain them all together in one chain ? Your parameters would be
```
--led-rows=16 --led-cols=32 --led-chain=5
```
Depending on if these are outdoor panels or not, you might need to have to choose a`--led-multiplexing=xxx`parameter.

Given that you have a the LED panels not in their natural orientation (which is over the 32 width, and clocked from a Pi on the right), you need to [write a PixelMapper](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates) that does the coordinate mapping.

ostium, 2018-03-08:
My setup is this: 80 columns and 32 lines:
[image]

but, if I execute this, the image is not in landscap format:
sudo ./demo -t 10 -D 1 runtext.ppm --led-gpio-mapping=classic-pi1 --led-rows=16 --led-cols=32 --led-chain=5

[image]

Maybe is a rotational problem on software? Where I can change this?

Thanks

ostium, 2018-03-08:
Sorry, I didn't read all off your response... "you need to write a PixelMapper that does the coordinate mapping."

angelogoncalve, 2018-03-11:
Use this transformer to 32x16 panel:

void SArrangementTransformer::TransformCanvas::SetPixel(
  int x, int y, uint8_t red, uint8_t green, uint8_t blue) {
  if (x < 0 || x >= width_ || y < 0 || y >= height_) return;
  const int slab_height = virtualrows_*panel_height_;   // one folded s-shape
  const int base_y = (y / slab_height) * panel_height_;
  y %= slab_height;
   if (y < panel_height_) {
   x += delegatee_->width()*2/3 ;

     }
 if ((y < 2*panel_height_ ) && (y >= panel_height_)) {
    x = 2*width_ - x - 1;
    y = slab_height *2/3 - y - 1;

  }

 if  ((y < 3*panel_height_ ) && (y >=2* panel_height_)) {
    
     y =  y - slab_height*2/3 ;

  }
  delegatee_->SetPixel(x, base_y + y, red, green, blue);
}

angelogoncalve, 2018-03-11:
You can try to rotate the image doing this function:
    
def rotate_image(self, image, angle):
return image.convert("RGB").rotate(angle)

ostium, 2018-07-13:
@angelogoncalve 

Hello, thank you for the help... your #611 save my day! It works very well!

closed 2025-11-27 by marcmerlin