hzeller/rpi-rgb-led-matrix #553: Pixel Quality of Img (led-slowdown-gpio)

dma19, 2018-03-11:
Hello,

I have pixels that appear abnormally with my rpi 3,
--led-rows=32 --led-cols=32 --led-chain=2  --led-parallel=2  --led-multiplexing=2

When I leave it by default this option: --led-slowdown-gpio=1  :
[image]

Becomes better with: --led-slowdown-gpio=2 :
[image]

But not top :( have pixel problem, false position/color? don't understand, I think the problem come from cpu/ressource of rpi 3?

Thank you!

angelogoncalve, 2018-03-11:
Try this first option (maybe the second option can also help):

--led-slowdown-gpio=4
--led-pwm-bits=1

hzeller, 2018-03-17:
Do you have a logic level converter ? If the position is shifted, I could imagine that the clock signals are not properly registered by the LED panel, which can happen if you directly connect the 3.3V logic level of the Pi to the inputs of the panels that expect 5V logic.

Also make sure that the GND of your Pi is properly connected to the GND of the panel - not doing that can sometimes accidentally work, but could result in similar artifacts.