hzeller/rpi-rgb-led-matrix #554: Issue with pixels not mapping properly

jantznick, 2018-03-13:
First let me give a huge thank you for putting this together and being so helpful in the issues. I am trying to get my 32x16 LED matrix running properly and it seems the pixels are not being displayed properly.

When I run this command:
sudo ./examples-api-use/demo -D 0 --led-no-hardware-pulse --led-rows=16

This is my output:
https://youtu.be/-FpBQtVt3Qo

I also tried running it with --led-multiplexing=4 and this was the result:
https://youtu.be/KP74P1oB0f4

Any help in this would be greatly appreciated!

hzeller, 2018-03-13:
None of `--led-multiplexing` values (from 1 to 6) did give a usable result ?

jantznick, 2018-03-13:
No, 4 gives the closest, but I can do them over again tomorrow evening and upload video of them if that would help.

hzeller, 2018-03-13:
You problably have to go systematically through the pixels and see where they end up, then write a new [MultiplexMapper](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc#L119) that indicates where each pixel goes.

Probably easiest is to start out with an existing one and adapt it to your needs.

NeerajShrivastava04, 2018-03-13:
Can you please try to change the below line of code in ZStripeTransformer in multiplex-mappers.cc file.

const int vert_block_is_odd = (((y+4) / tile_height) % 2);

Over here calculation of vert_block_is_odd is being modified 4 is being with the value of before dividing it with tile_height.

jantznick, 2018-03-13:
Yes, I will try both of these when I get home tonight. Thank you again so much for your help and work on this.

vapiper, 2018-03-13:
This expression `vert_block_is_odd = (((y+4) / tile_height) % 2);`  doesn't make any sense, because you just invert the result  of the original expression. 
You can get the same result by adding the new instance of the `ZStripeMultiplexMapper` class with swapped offset parameters:
```
  result->push_back(new ZStripeMultiplexMapper("MirrorZStripe", 8, 0));
```
Let me know about the result.

angelogoncalve, 2018-03-13:
Try to use:

sudo ./examples-api-use/demo -D 0 --led-no-hardware-pulse --led-rows=16 --led-multiplexing=4 --led-row-addr-type=2

or

sudo ./examples-api-use/demo -D 0 --led-no-hardware-pulse --led-rows=16 --led-multiplexing=5

or try to modify the ZStripeTransformer in multiplex-mappers.c:

void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
static const int tile_width = 8;
static const int tile_height = 4;

const int vert_block_is_odd = (((y+4) / tile_height) % 2);  //((y / tile_height) % 2);

const int even_vblock_shift = (1 - vert_block_is_odd) * even_vblock_offset_;
const int odd_vblock_shitf = vert_block_is_odd * odd_vblock_offset_;

//const int xoffset =8 * (x / 8);
//const int yoffset = ((y+4) % 8) / 4;

//*matrix_x = x  + 8 * yoffset + xoffset;
//*matrix_y = (y % 4) + 4 * (y / 8);

*matrix_x = x + (((x+even_vblock_shift)/tile_width) * tile_width) + odd_vblock_shitf;
*matrix_y = (y % tile_height) + tile_height * (y / (tile_height * 2));

}`

jantznick, 2018-03-15:
I tried all of the above options, and they didn't seem to change the output much from just using --led-multiplexing=4. 

I would like to go through and write a new MultiplexMapper, but am not sure where to start, is there a quick overview I can read over before diving into that? I suppose I will start by looking through some of the previous issues and seeing if I can't decode anything from that.

hzeller, 2018-03-17:
The general thing that is happening is, that the multiplexed boards internally are not the size the are on the outside, like `32x16`, but internally, they are more like `64x8`, in which the pixels are folded and remapped. So what we have to do is to map a coordinate in the 32x16 range to the 64x8 range.

This is what the Multiplex mappers are doing. They have one method 
```
void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y)
```
that gets the input coordinates (x, y), manipulates them in some way and then assigns the results to the `*matrix_x` and `*matrix_y` values. With that in mind, you can read what each of the mappers is doing in the [lib/multiplex-mappers.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc) file.

The pixel-mapper that is closes to what you need is the [ZStripeMultiplexMapper](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc#L137), so I suggest to start manipulate the code there and test.

If you have a code that works, please post it here, so that we then can add it as a new permanent addition to the multiplex mappers.

angelogoncalve, 2018-03-17:
Try this:

# The following code converts a 16x32 matrix into a 64x8 matrix for proper display

   new_x = x
   new_y = y
   if y < 4:
      new_x = x + 32
   elseif y < 8: 
      new_y = y - 4
   elseif y < 12: 
   
      new_x = x + 32
      new_y = y - 4
   
   else: 
      new_y = y - 8

Puneetkumarpal, 2018-04-23:
I am facing same problem. please help if there is any solution.

angelogoncalve, 2018-04-23:
Try the option:

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-multiplexing=7 --led-row-addr-type=0

Puneetkumarpal, 2018-04-24:
it worked when i change in [multiplex-mappers.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/212b0bbccbbd9adacbf0d49b6c0adbaeb7e83422/lib/multiplex-mappers.cc#L209) 
  result->push_back(new ZStripeMultiplexMapper("ZStripe", 0, 8)); to   result->push_back(new ZStripeMultiplexMapper("ZStripe", 8, 0));

closed 2025-11-27 by marcmerlin