hzeller/rpi-rgb-led-matrix #559: Problems with P5 32x32 1/8 scan display.

Lookandfind, 2018-03-17:
Hi guys thanks for work for this library. 

I read that other people had simular problems with first use, but with their solution i couldn't solve my problem.
I wired the display after the schematic on the first page and tryed a lot of different parameters like --led-multiplexing=0..5
But the best result is with multiplex =0
As you can see in the picture, i  belive that i have some lines double?
Can anybody give me a hint what i can do?

thanks for your time 

[image]

hzeller, 2018-03-18:
Do you have a picture of the connector on the board, that might indicate further details about the address mode ? Do you have a link or description of hte particular panel ?

The double lines indicate that it might be something like `--led-row-addr-type=2`, but then with an unusual case of 1:4 multiplexing ? Maybe it has a yet unknown address type format.

angelogoncalve, 2018-03-18:
apply --led-multiplexing=6

or --led-multiplexing=4 --led-row-addr-type=2

or

MATRIX_ROWS = 32
MATRIX_COLUMNS = 32

    def chinese(x, y):
        scan_rate = MATRIX_ROWS / 4

            odd8_block = (((y % (.MATRIX_ROWS / 2)) / scan_rate) + 1) % 2
            num_mod_x = x / MATRIX_COLUMNS
            new_x = x + MATRIX_COLUMNS * (num_mod_x + odd8_block)
            new_y = (y % scan_rate) + scan_rate * (y / (MATRIX_ROWS / 2))

        return new_x, new_y

or

MATRIX_ROWS = 32
MATRIX_COLUMNS = 32

    def electrodragon(x, y):
            scan_rate = MATRIX_ROWS / 4
            new_x = ((x / (MATRIX_COLUMNS / 2)) * MATRIX_COLUMNS) + (x % (
                        MATRIX_COLUMNS / 2))
            if (y / scan_rate) % 2 == 1:
                new_x += MATRIX_COLUMNS / 2
            new_y = ((y / (MATRIX_ROWS / 2)) * (MATRIX_ROWS / 2)) + (y % scan_rate)
        return new_x, new_y

Lookandfind, 2018-03-19:
Hi, 
I dont have further informations about the display i bought it over a yeahr ago on Aliexpress.
But here is a picture of the back

[image]

This is a pic of using --led-multiplexing = 6

closed 2018-03-19 by Lookandfind

rafayhayat, 2021-04-07:
I also have 32*32 panel with 1/8 scan rate, could you please suggest me any of the library as adafruit RGB Matrix Panel library is not working with the panel