hzeller/rpi-rgb-led-matrix #562: Is the -l option (logarithmic mapping) deprecated?

btownshend, 2018-03-19:
Seems that the code now marks the -l option as "Hidden option. Still supported but no really useful", and defaults the mapping to linear.    It seems to me that the logarithmic CIE mapping is still quite useful.

hzeller, 2018-03-19:
No, the logarithmic mapping is the default.
The linear mapping doesn't really make sense in most of the cases (because that is not how our eyes work), so this is why it is not really prominently advertised.

btownshend, 2018-03-19:
It looks to me like the default is linear.   See pixel-push.cc:

99: pp_options.is_logarithmic = false;
181: matrix->set_luminance_correct(pp_options.is_logarithmic);

Brent

hzeller, 2018-03-19:
Ah, that default is, because many people dealing with LED strips (which is the PixelPusher primary audience) are dealing with linear strips. But yeah, given that the default is the 'wrong' value, it should not be a hidden option.

hzeller, 2018-03-19:
Fixed in https://github.com/hzeller/rpi-matrix-pixelpusher/commit/108ddf19d4c54f9682b5259a0b1bda32f9894363

closed 2018-03-19 by hzeller