hzeller/rpi-rgb-led-matrix #563: [Large Screen] 64x64 Matrix

DnR-iData, 2018-03-19:
Hi mates,

I would like to know how I can in Python set the Large Screen Mode, like the "**-L**" argument with the demo command : 
```bash
sudo ./demo -D0 --led-rows=32 --led-cols=32 --led-chain=4 -L --led-brightness=30
```
[image]

The second part of the "O" is missing with Python :
[image]

I don't really understand how transformers works.

Can you give me a little help please.

Thank You in advance.

iData

hzeller, 2018-03-19:
You need to update to the latest version of this library, then use the flags
  `--led-pixel-mapper="U-mapper" --led-chain=4`
This will work for this arrangement in both, C++ and Python programs.
If you are working with the options directly in your program, set the `pixel_mapper_config` value to `"U-mapper"`.

DnR-iData, 2018-03-19:
Perfect :
[image]

Thank you so mutch.

All ok :)

DnR-iData, 2018-03-19:
I close the issu.

Thx

closed 2018-03-19 by DnR-iData