hzeller/rpi-rgb-led-matrix #564: Problems with P5 32x32 1/8 scan display

Lookandfind, 2018-03-19:
#559 
Sorry I'm new to github, so i closed accidentally the Issue. 

Sorry for that.

angelogoncalve, 2018-03-19:
Try apply this options:

--led-multiplexing=1

--led-multiplexing=6

or --led-multiplexing=4 --led-row-addr-type=2

or

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

options = RGBMatrixOptions()
number_of_panels = 1
MATRIX_ROWS = 32
MATRIX_COLUMNS = 32
options.rows = MATRIX_ROWS / 2
options.cols = MATRIX_COLUMNS
options.chain_length = 2 * number_of_panels
display = RGBMatrix(options=options)
height = display.height
width = display.width

    def chinese(x, y, red, green, blue):
        scan_rate = MATRIX_ROWS / 4

        odd8_block = (((y % (MATRIX_ROWS / 2)) / scan_rate) + 1) % 2
        num_mod_x = x / MATRIX_COLUMNS
        new_x = x + MATRIX_COLUMNS * (num_mod_x + odd8_block)
        new_y = (y % scan_rate) + scan_rate * (y / (MATRIX_ROWS / 2))
        return display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, width):
    for y in range(0, height):
        chinese(x, y, 255, 0, 0)
        print '({}, {})'.format(x, y)
        time.sleep(1)

or

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

options = RGBMatrixOptions()
number_of_panels = 1
MATRIX_ROWS = 32
MATRIX_COLUMNS = 32
options.rows = MATRIX_ROWS
options.cols = MATRIX_COLUMNS
options.chain_length = 2 * number_of_panels
display = RGBMatrix(options=options)
height = display.height
width = display.width

    def electrodragon(x, y, red, green, blue):
        scan_rate = MATRIX_ROWS / 4
        new_x = ((x / (MATRIX_COLUMNS / 2)) * MATRIX_COLUMNS) + (x % (MATRIX_COLUMNS / 2))
        if (y / scan_rate) % 2 == 1:
            new_x += MATRIX_COLUMNS / 2
        new_y = ((y / (MATRIX_ROWS / 2)) * (MATRIX_ROWS / 2)) + (y % scan_rate)
        return display.SetPixel(new_x, new_y, red, green, blue)

   for x in range(0, width):
       for y in range(0, height):
           electrodragon(x, y, 0, 255, 0)
           print '({}, {})'.format(x, y)
           time.sleep(1)

Lookandfind, 2018-03-19:
Hi, 
I dont have further informations about the display i bought it over a yeahr ago on Aliexpress.
But here is a picture of the back

[image]

This is a pic of using --led-multiplexing = 6
[image]

This is with  --led-multiplexing=4 --led-row-addr-type=2
[image]

This is an error i get when i try to run the pythonscript:
[image]

angelogoncalve, 2018-03-19:
I fix the bug above in the code, see please.

Lookandfind, 2018-03-19:
This is what i get:
[image]

angelogoncalve, 2018-03-19:
for x in range(0, width):   
      for y in range(0, height):   
         electrodragon(x, y, 0, 255, 0)   
         print '({}, {})'.format(x, y)   
         time.sleep(1)

Lookandfind, 2018-03-19:
Still the same error. 
[image]

Lookandfind, 2018-03-19:
Thanks for your help. Now its working. 
[image]