hzeller/rpi-rgb-led-matrix #565: Is this Chinese P10RGB outdoor module compatible ?

vegetablesalad, 2018-03-19:
I'm trying to get this module to work, but its jumbled up, I think I have tried every setting.
Is it supported or it has some weird mapping I have to transform myself?
Just want to check before committing onto remapping.

[P10RGB doc PDF](http://www.sanderled.com.tw/english/sdm-p10rgb-osab1s-3216.pdf)

angelogoncalve, 2018-03-19:
Try apply this options:

--led-multiplexing=1

or 

--led-multiplexing=4 --led-row-addr-type=2

or

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

options = RGBMatrixOptions()
number_of_panels = 1
MATRIX_ROWS = 16
MATRIX_COLUMNS = 32
options.rows = MATRIX_ROWS / 2
options.cols = MATRIX_COLUMNS
options.chain_length = 2 * number_of_panels
display = RGBMatrix(options=options)
height = display.height
width = display.width

    def chinese(x, y, red, green, blue):
        scan_rate = MATRIX_ROWS / 4

        xoffset = (MATRIX_COLUMNS / 4) * (x / (MATRIX_COLUMNS / 4))
        yoffset = ((y + scan_rate) % (.MATRIX_ROWS / 2)) / scan_rate
        new_x = x + (MATRIX_COLUMNS / 4) * yoffset + xoffset
        new_y = (y % scan_rate) + scan_rate * (y / (MATRIX_ROWS / 2))
        return display.SetPixel(new_x, new_y, red, green, blue)

for x in range(0, width):
    for y in range(0, height):
        chinese(x, y, 255, 0, 0)
        print '({}, {})'.format(x, y)
        time.sleep(1)

vegetablesalad, 2018-03-20:
Thanks, will try this tonight!

hzeller, 2018-03-27:
Did it work ? If not, you probably have to go through the datasheet and build a [multiplex mapper](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#multiplex-mappers) and possibly send a pull request.

vegetablesalad, 2018-03-28:
I managed to get it somewhat close. I'll post my results tonight or later this week.

vegetablesalad, 2018-05-22:
Sorry didn't have any free time to work on my project, but just picked it up.
So I tried quite a few combinations of settings and your examples, but can't quite get it working.
Here are some of the things I tried:

- from all the demos this setup got me the closest result:
> sudo ./demo -D 0 --led-no-hardware-pulse --led-chain=1 --led-rows=16 --led-multiplexing=5 --led-row-addr-type=2
result: https://youtu.be/HAGbXrgpqEs

- Your first python [code sample](https://gist.github.com/vegetablesalad/df5f2d23776e4b70ba5d44b510902852) 
result: https://youtu.be/72ZKh3QaEw4

- And final python [code](https://gist.github.com/vegetablesalad/35fd40f9ba5a05f834d4f55ac655c5fb)
result: https://youtu.be/Wj29KVa5SuI

doesn't look like I'm close, this panel must be quite different

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin