hzeller/rpi-rgb-led-matrix #566: Problem when call graphics function rgbmatrix library in Python

angelogoncalve, 2018-03-19:
Anybody has problems when call in Python the library rgbmatrix:

from rgbmatrix import graphics

Can you help me Dear Henner Zeller please? Thank you for your attention and help my friend.

hzeller, 2018-03-19:
Did you [install the python bindings](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python#python-2) and tested the examples in [`bindings/python/samples`](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python/samples) ? If the examples don't work, the installation was not successful.
If the examples work, then it should be possible to use in your own code.

angelogoncalve, 2018-03-20:
I have this error when I test the import graphics from rgbmatrix library running the Python file runtext.py using RPi3 + Raspbian Stretch + 32x32 matrix + hat. 

How can fix please Dear Henner Zeller. The issue 567 "ValueError: rgbmatrix.core.RGBMatrixOptions - try recompiling" has the same problem. 

Thank you for your attention and help Dear friend.

Traceback (most recent call last):
  File "runtext.py", line 4, in <module>
    from rgbmatrix import graphics
  File "rgbmatrix/core.pxd", line 12, in init rgbmatrix.graphics (rgbmatrix/graphics.cpp:2842)
ValueError: rgbmatrix.core.RGBMatrixOptions has the wrong size, try recompiling. Expected 92, got 88

hzeller, 2018-03-20:
so, did you do what the error message suggested, to , make clean and recompile?

angelogoncalve, 2018-03-20:
Yes I do:

sudo make clean

sudo HARDWARE_DESC=adafruit-hat make install-python

or it is need also to:

sudo apt-get update && sudo apt-get install python2.7-dev python-pillow -y
make build-python

?

or it is another type of problem to fix?

Thank you for your help Dear Henner Zeller.

hzeller, 2018-03-20:
Looks like a simple link problem then. Can you fix and send a pull request?
Also, if you're not entirely sure if the make clean worked, a fresh clone is also a good starting point.

angelogoncalve, 2018-03-20:
No I can not fix because I do not know how can fix this error? 

Yes I remove the folder and the files in the dist/utils from the raspberry that haded created when install the library.

Yes I also used a fresh clone also and did not work too the graphics for example with runtext.py file:

git clone https://github.com/hzeller/rpi-rgb-led-matrix
cd rpi-rgb-led-matrix
sudo HARDWARE_DESC=adafruit-hat make install-python

Did you know why appears this problem because before the pixelmapper branch the library was working fine also for graphics. 

Now only works RGBMatrix and RGBMatrixOptions not graphics for loading bdf fonts to write a text in a panel.

Thank you for your attention and help Dear Zeller.

angelogoncalve, 2018-03-20:
I have some difficult with C code also. I only know something in Python language. 

Sorry my Dear friend I want to help but I do not know why the problem comes if I remove all from the rgbmatrix library and creat a fresh clone new one and the error for graphics type persist the same but if you put in the compiler from rgbmatrix import graphics not appears an error.

angelogoncalve, 2018-03-20:
The error comes from the file graphics.cpp Dear friend Zeller. Is a way to fix it please? Thank you for your help my friend.
 
(rgbmatrix/graphics.cpp: line 2842)

 __pyx_ptype_9rgbmatrix_4core_RGBMatrixOptions = __Pyx_ImportType("rgbmatrix.core", "RGBMatrixOptions", sizeof(struct __pyx_obj_9rgbmatrix_4core_RGBMatrixOptions), 1); if (unlikely(!__pyx_ptype_9rgbmatrix_4core_RGBMatrixOptions)) __PYX_ERR(1, 12, __pyx_L1_error)

/* TypeImport */
  #ifndef __PYX_HAVE_RT_ImportType
#define __PYX_HAVE_RT_ImportType
static PyTypeObject *__Pyx_ImportType(const char *module_name, const char *class_name,
    size_t size, int strict)
{
    PyObject *py_module = 0;
    PyObject *result = 0;
    PyObject *py_name = 0;
    char warning[200];
    Py_ssize_t basicsize;
#ifdef Py_LIMITED_API
    PyObject *py_basicsize;
#endif
    py_module = __Pyx_ImportModule(module_name);
    if (!py_module)
        goto bad;
    py_name = __Pyx_PyIdentifier_FromString(class_name);
    if (!py_name)
        goto bad;
    result = PyObject_GetAttr(py_module, py_name);
    Py_DECREF(py_name);
    py_name = 0;
    Py_DECREF(py_module);
    py_module = 0;
    if (!result)
        goto bad;
    if (!PyType_Check(result)) {
        PyErr_Format(PyExc_TypeError,
            "%.200s.%.200s is not a type object",
            module_name, class_name);
        goto bad;
    }
#ifndef Py_LIMITED_API
    basicsize = ((PyTypeObject *)result)->tp_basicsize;
#else
    py_basicsize = PyObject_GetAttrString(result, "__basicsize__");
    if (!py_basicsize)
        goto bad;
    basicsize = PyLong_AsSsize_t(py_basicsize);
    Py_DECREF(py_basicsize);
    py_basicsize = 0;
    if (basicsize == (Py_ssize_t)-1 && PyErr_Occurred())
        goto bad;
#endif
    if (!strict && (size_t)basicsize > size) {
        PyOS_snprintf(warning, sizeof(warning),
            "%s.%s size changed, may indicate binary incompatibility. Expected %zd, got %zd",
            module_name, class_name, basicsize, size);
        if (PyErr_WarnEx(NULL, warning, 0) < 0) goto bad;
    }
    else if ((size_t)basicsize != size) {
        PyErr_Format(PyExc_ValueError,
            "%.200s.%.200s has the wrong size, try recompiling. Expected %zd, got %zd",
            module_name, class_name, basicsize, size);
        goto bad;
    }
    return (PyTypeObject *)result;
bad:
    Py_XDECREF(py_module);
    Py_XDECREF(result);
    return NULL;
}
#endif

hzeller, 2018-03-20:
Can you do a fresh `git pull` and recompile the Python lib and see if it works ?

closed 2018-03-20 by hzeller

angelogoncalve, 2018-03-21:
Yes it works my friend, thank you for your help Dear Zeller.

commit 3f63f54e (2018-03-20): Re-generate graphics cpp.

Potentially fixes #566 #567