hzeller/rpi-rgb-led-matrix #567: ValueError: rgbmatrix.core.RGBMatrixOptions - try recompiling

adilberrada, 2018-03-20:
Hi,
Thanks a lot for this great library! I hope you could help as I am having two problems:

- RPi3 + Raspbian jessie + 64x32 matrix + hat: no problem with the matrix but when I add a 3G dongle on my pi, it freezes after a while... any idea?
- RPi3 + Raspbian Stretch + 64x32 matrix + hat: I thought my problem with the 3G dongle was maybe due to an old version of the OS so I updated to Raspbian Stretch from scratch. I cloned the git directory and built the project but when I execute the example runtext.py (or my script) I got the following issue and I have no clue how to fix it...
------------------------------------------------------------------
File "runtext.py", line 4, in <module>
from rgbmatrix import graphics
File "rgbmatrix/core.pxd", line 12, in init rgbmatrix.graphics (rgbmatrix/graphics.cpp:2842)
ValueError: rgbmatrix.core.RGBMatrixOptions has the wrong size, try recompiling. Expected 92, got 88
-----------------------------------------------------------------

Some help would be VERY appreciated!

Thanks a lot community ;)

angelogoncalve, 2018-03-20:
I have the same error when I tested the import graphics from rgbmatrix library running the Python file runtext.py using RPi3 + Raspbian Stretch + 32x32 matrix + hat + ethernet cable LAN.

Dear friend if you know how can be fixed please share. 

Thank you for your attention and help Dear friend.

Traceback (most recent call last):
File "runtext.py", line 4, in
from rgbmatrix import graphics
File "rgbmatrix/core.pxd", line 12, in init rgbmatrix.graphics (rgbmatrix/graphics.cpp:2842)
ValueError: rgbmatrix.core.RGBMatrixOptions has the wrong size, try recompiling. Expected 92, got 88

hzeller, 2018-03-20:
Can you do a fresh `git pull` and recompile the Python lib and see if it works ?

closed 2018-03-21 by hzeller

angelogoncalve, 2018-03-21:
Yes it works fine Dear Zeller, thank you for your attention my friend.