hzeller/rpi-rgb-led-matrix #568: Bad performance in own project, works in this repo

iUltimateLP, 2018-03-20:
Hey,
I'm having a very weird problem which I'm not sure the origin of. I'm writing my custom C++ driver using this library together with a WebSocket library so I am able to control the display from my smartphone. That's all good, but I'm having very strange performance issues. For testing, I copied the RotatingBlockGenerator out of the demo, and set it up in my code. But when I start it, the cube is rotating really really slow, not fluid at all. If I run it directly from the demo code of this repo, it works fluid. So it has to do with my project, but I can't find the issue. I disabled everything additional which came from my previously written code, so it is just the RotatingBlockGenerator class, and it's creating that in the main entry point, but still - bad performance. The code:
```cpp
bool b = false;

// Termination signal handler
void SignalHandler(int Signal)
{
    printf("\n");
    Logging::Log(ELogCategory::LogMainframe, ELogVerbosity::Message, "Caught signal %d", Signal);
    b = true;
    // Stops the web socket server
    //WebSocketServer::Get()->StopServer();
}

// Simple class that generates a rotating block on the screen.
class RotatingBlockGenerator : public rgb_matrix::ThreadedCanvasManipulator {
public:
  RotatingBlockGenerator(rgb_matrix::Canvas *m) : rgb_matrix::ThreadedCanvasManipulator(m) {}

  uint8_t scale_col(int val, int lo, int hi) {
    if (val < lo) return 0;
    if (val > hi) return 255;
    return 255 * (val - lo) / (hi - lo);
  }

  void Run() {
    const int cent_x = canvas()->width() / 2;
    const int cent_y = canvas()->height() / 2;

    // The square to rotate (inner square + black frame) needs to cover the
    // whole area, even if diagnoal. Thus, when rotating, the outer pixels from
    // the previous frame are cleared.
    const int rotate_square = std::min(canvas()->width(), canvas()->height()) * 1.41;
    const int min_rotate = cent_x - rotate_square / 2;
    const int max_rotate = cent_x + rotate_square / 2;

    // The square to display is within the visible area.
    const int display_square = std::min(canvas()->width(), canvas()->height()) * 0.7;
    const int min_display = cent_x - display_square / 2;
    const int max_display = cent_x + display_square / 2;

    const float deg_to_rad = 2 * 3.14159265 / 360;
    int rotation = 0;
    while (running()) {
      ++rotation;
      usleep(0 * 1000);
      rotation %= 360;
      for (int x = min_rotate; x < max_rotate; ++x) {
        for (int y = min_rotate; y < max_rotate; ++y) {
          float rot_x, rot_y;
          Rotate(x - cent_x, y - cent_x,
                 deg_to_rad * rotation, &rot_x, &rot_y);
          if (x >= min_display && x < max_display &&
              y >= min_display && y < max_display) { // within display square
            canvas()->SetPixel(rot_x + cent_x, rot_y + cent_y,
                               scale_col(x, min_display, max_display),
                               255 - scale_col(y, min_display, max_display),
                               scale_col(y, min_display, max_display));
          } else {
            // black frame.
            canvas()->SetPixel(rot_x + cent_x, rot_y + cent_y, 0, 0, 0);
          }
        }
      }
    }
  }

private:
  void Rotate(int x, int y, float angle,
              float *new_x, float *new_y) {
    *new_x = x * cosf(angle) - y * sinf(angle);
    *new_y = x * sinf(angle) + y * cosf(angle);
  }
};

// Main entry point for this application
int main(int argc, char** argv)
{
    Logging::Log(ELogCategory::LogMainframe, ELogVerbosity::Message, "LED Matrix driver mainframe started");

    // Register the SIGINT handler
    signal(SIGINT, SignalHandler);

    rgb_matrix::RGBMatrix::Options defaults;
    defaults.hardware_mapping = "regular";
    defaults.rows = 64;
    defaults.chain_length = 2;
    defaults.parallel = 1;
    //defaults.scan_mode = 0;
    
    rgb_matrix::RuntimeOptions roptions;

    rgb_matrix::RGBMatrix* RgbMatrix = rgb_matrix::CreateMatrixFromOptions(defaults, roptions);

    rgb_matrix::Canvas* canvas = RgbMatrix;

    rgb_matrix::ThreadedCanvasManipulator* ImageGen = new RotatingBlockGenerator(canvas);
    ImageGen->Start();

    while (!b)
    {
// Just wait
    }

    Logging::Log(ELogCategory::LogMainframe, ELogVerbosity::Message, "Mainframe requested to exit!");

    return 0;
}
```
Any idea why this could have this bad performance?
Dankeschön :smile:

hzeller, 2018-03-20:
When you compile your own project: did you use all the same compile options, in particular the `-O3` optimize option ?

iUltimateLP, 2018-03-20:
Damn yeah, that fixed it. For my understanding, what does -O3 do?

hzeller, 2018-03-20:
`-O` is the option to tell the compiler what level of optimization the compiler should apply. `-O0` would just generate some very simple code (but that is easy to debug), while `-O1`, `-O2` and `-O3` supply increasing ninja-levels of optimization when generating the code. It takes longer to compile, but generates faster code; which is almost always what you want in the final production code.

closed 2018-03-20 by hzeller

iUltimateLP, 2018-03-22:
Hey again, I tried to run the rotating cube example, and it looks good, but it still is a bit slower than it is in the original demo code, and from this point I'm not sure why. I'm looking to increase the performance to a maximum possible, and this grinds my gears. The code is the same, I just have a Display wrapper class which contains a TestAnim() function which runs that, but extra wrapper classes shouldn't break the performance 😢