hzeller/rpi-rgb-led-matrix #572: Pi 3 B+

SparkyXI, 2018-03-22:
Upgraded to a 3B+ (including updating the OS to latest), and now my matrix seems dead in the water.

Does this repo need compatibility updates to allow for the 3B+?

hzeller, 2018-03-22:
So I did a brief test with a single chain the 3B+ and the latest Raspbian and it seems to work. So I didn't do very thorough tests yet, but at least the first test seems to work.

How do you have the matrix connected, do you have logic level converters ? How many chains ?

christopherbunnell, 2018-03-24:
I am perhaps having the same issue. 3B+ connected to a single 32x16 RGB LED Matrix. Will the demo run on this or do I need to tell it that I have a single 32x16, and if so, how do I give that instruction?

SparkyXI, 2018-03-27:
I did figure this out - I had to rebuild the repo from scratch... now I'm trying to figure out my defaults again. I'm not the best with this, but I'll get there. ;)

closed 2018-03-27 by SparkyXI

hzeller, 2018-03-27:
Cool. I suggest to configure everything with the command line flags, so that is easier to remember next time you have to upgrade the repo.

SparkyXI, 2018-03-27:
Interesting - could you please give me an example of this? For instance, I'm (again) having a hard time finding how to set 16x32 as default...

If what you say is what I think you say, I can update the repo whenever I want, then just add flags when I make?

hzeller, 2018-03-27:
Even simpler: you can give the command line options when you run your program if you are using `CreateMatrixFromOptions()`. No changes at compile time needed.

You can give the command line flags `--led-rows=16` on your program (see the `examples-api-use/demo` program), so that you select the 16 panel. Chain length is `--led-chain=1` and so on.

The are a [bunch of other flags](https://github.com/hzeller/rpi-rgb-led-matrix#changing-parameters-via-command-line-flags) that allow tweaking.