hzeller/rpi-rgb-led-matrix #574: Error when displaying samples on LED Matrix

kjb05, 2018-03-23:
Hi, I've tried a couple different samples but every time I run them i get this output on my board:
[image]

Any ideas?

Thanks in advance for any help!

angelogoncalve, 2018-03-23:
Try do this:

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 32
number_of_panels = 2
parallel = 1
number_of_columns = 32

options = RGBMatrixOptions()

def rgbmatrix_options():
    #options = RGBMatrixOptions()
    options.multiplexing = 1
    options.row_address_type = 0
    options.brightness = 100
    options.rows = number_of_rows / 1
    options.cols = number_of_columns
    options.chain_length = 1 * number_of_panels
    options.parallel = parallel
    options.hardware_mapping = 'regular'
    options.inverse_colors = False
    options.led_rgb_sequence = "RGB"
    options.gpio_slowdown = 4
    options.pwm_lsb_nanoseconds = 130
    options.show_refresh_rate = 0
    options.disable_hardware_pulsing = True
    options.scan_mode = 1
    options.pwm_bits = 1
    options.daemon = 0
    options.drop_privileges = 0
    options.pixel_mapper_config = "Rotate:0"
    
    return RGBMatrix(options=options)

display = rgbmatrix_options()

height = display.height
width = display.width

for x in range(0, width):
    for y in range(0, height):
        
        display.SetPixel(x, y, 0, 255, 0)
        
        point = Point(x, y)
        print point
        
        time.sleep(1)

hzeller, 2018-03-23:
Check that address-line `C` is properly connected. It seems to be clamped to ground.

kjb05, 2018-03-24:
Sorry I am totally new to this, what do you mean by address line c?

hzeller, 2018-03-24:
The line `C` as described in the wiring diagram:
  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#wiring-diagram
If you are using an adapter, check if everything from GPIO pin 18 to the chip and from there to the display (PCB and wiring) is working.

kjb05, 2018-03-24:
I dont have any tools to actually test this the correct way but I checked to make sure everything was soldered correctly and it is. I fixed a pin that might have had a poor connection and when i tried to run another sample i got this
[image]

The same thing just with different colors. Can you think of anything else that may be causing this besides a bad GPIO connection?

hzeller, 2018-03-24:
What are the current commandline flags you give to the programs ?
Can you take a picture of the output of
```
 ./demo --led-cols=64 --led-rows=32 -D3
```
?

kjb05, 2018-03-24:
Is this what you mean?
[image]

hzeller, 2018-03-24:
No, in the `examples-api-use/` https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use

kjb05, 2018-03-24:
Ok, here is what i receive when running that commmandline:
[image]
[image]

hzeller, 2018-03-24:
It's uppercase `-D`

kjb05, 2018-03-24:
Ah, well here is the output for that 
[image]

Nothing was displayed on the LED Board

hzeller, 2018-03-24:
you have compiled the library with adafruit hat. Is this the adapter you are using?

kjb05, 2018-03-24:
Yes - this product https://www.adafruit.com/product/2345

hzeller, 2018-03-24:
ok, so starting from here, you need to make it work,  e.g. check power, make sure to have the right address mode and multiplexing etc. The readmed help with all the details. Make sure to use --Led-rows _this_ library and read this readme.

closed 2025-11-27 by marcmerlin