hzeller/rpi-rgb-led-matrix #576: Compatability Issues

Grimmy347, 2018-03-30:
i was trying to use the adafruit web interface demo but I could not get it to work with my 64x32 matrix.  I updated to your most current database but then the web interface no longer works.  Any ideas how I can make your repository work with the webserver by adafruit?  

this is the link to theirs.  thank you!!
https://learn.adafruit.com/raspberry-pi-rgb-led-matrix-webapp/software

Grimmy347, 2018-03-31:
Should I be able to just use the adapter file from your distro and copy it over to the one in the older distro?

https://github.com/onebeartoe/rpi-rgb-led-matrix

hzeller, 2018-03-31:
Please ask Adafruit w.r.t. the web interface, I have not looked at that code to comment on it. Probably, they just need to update the dependency submodule.