hzeller/rpi-rgb-led-matrix #577: Scrolling display jitters

chollan, 2018-04-01:
Hi there, I see this is very active in responses and don’t know where else to post my question so here’s hoping: 

I have installed and am running your examples very successfully, but I’ve run into a few issues, namely with the chained displays. I’m chaining 3 16x32 displays and the display closest to the rpi is the clearest, and the other 2 have pretty bad jitter. I have tried to rewire multiple times thinking it was the wiring but it always turns out the same. 

Here’s a video example 
https://youtu.be/H0WVJKEG1Fk

If this isn’t the right channel to ask these questions in, can you direct me there?  Thanks in advance!  This is a lot of great work here!

hzeller, 2018-04-01:
Try to increase the `--led-slowdown-gpio` value to 2 or 3 or so.

I think it is worthwhile to consider adding a voltage level converter, maybe the digital input stages of the panels can't exactly trigger on the 3.3V digital levels coming from the Pi.

chollan, 2018-04-01:
Thank you for your reply, I’ll take a look at it tomorrow when I’m able to. Just out of curiosity, is it not possible to use the 5v GPIO pin 2 or 4?  It may be a simple answer, but I’m not too familiar with the GPIOs so it’s just something I’m not aware of.

hzeller, 2018-04-01:
There is no GPIO that outputs 5V digital signal, they all are 3.3V.

Pin 2 and 4 are the 5V DC voltage coming from the USB connector which can be used to power a Pi hat (or, it is also possible to power the Pi through these pins).

chollan, 2018-04-01:
Again, excuse my lack of knowledge, Would it be more benificial to get the hat, or the signal converter?  I plan on adding more chains and from what I’ve aeen, the Adafruit hat for the rpi does the conversion on the hat itself.

hzeller, 2018-04-01:
The Adafruit hat does level conversion yes. But you can only have one chain.
If you build the adapter in this project, you can have up to three parallel chains https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3
Depends on how many displays you consider attaching if it is worthwhile.

depili, 2018-04-01:
There is also the built adapter from China:

http://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

I have thus far used 13 such boards and they seem to work. For some reason
there are also additional features on the this hat but I haven't used
them, as a cheap and inaccurate RTC isn't really useful for anything...

hzeller, 2018-04-01:
Yes, I have heard fhat this is working as well. They use the same wiring as my board apparently.

chollan, 2018-04-02:
Hi there, just an update: I was seeing the jitter when using the rotate:180 transformer.  Once i stopped using this transformer, the text appeared to scroll normally, however there's a lot of ghosting and if the brightness isn't at 100, the text gets garbled.  I took hzeller's advice above and ordered a rpi hat that does the voltage conversion. i'll update when it arrives.

hzeller, 2018-04-02:
The 180 degree turn using the rotate pixelmapper should not have any influence; I suspect the artifacts are content dependent and are more pronounced with the rotation.
Have you read the whole readme? There is a section about ghosting.

chollan, 2018-04-02:
Sorry, I wasn't implying that the rotate:180 transformer was the cause of the problem while it definitely sounded like that.  More happy that it wasn't as bad as it was once I no longer used the transformer.  I did read the entire readme, but the flags that are recommended and modifying the PWM didn't affect the ghosting.  However I was modifying these when using the rotate:180 transformer.  Let me try to modify these again tonight without the transformer and i'll update when i'm able.  Thanks for all your help!

chollan, 2018-04-03:
Hi there, quick update:  setting --led-slowdown-gpio to 3 worked for the examples given in the repo.  However when I attempt to compile my own applications, the jittering and severe ghosting is back.  My application is based off of a mix of the scrolling text and static text.  i'll try to do a little debugging on my own.

hzeller, 2018-04-03:
Make sure to compile your programs with `-O3` gcc optimization.
Also, if you do animations, consider double-buffering if you're not doing it already (`SwapOnVSync()`).

chollan, 2018-04-03:
Thanks for the tips. While I was doing some debugging I’m realizing that it’s the bottom row of text that’s causing the most problems. I attached a video as a demonstration. The commands I’m using are scrolling text demo without the -y parameter and then woth the -y parameter

https://youtu.be/GIxvPJfKP5E

hzeller, 2018-04-03:
How is your power supply ? Do you have a stable power supply, connected with short, thick cables to every connector, having all grounds properly connected together and have capacitors connected to the inlets of the panels ?

It is interesting that the lower line is in particular showing artifacts the more is shown on the boards, while the last 'test' text is scrolling through without much disturbance. This very much looks like a power problem.

chollan, 2018-04-03:
I have a 5V 10A switching power supply and used the cables that were sent With the matricies.  I also bought the Female DC Power adapter - 2.1mm jack to screw terminal block offered at Adafruit to connect the leads to the power supply.  There were no ground wires in the packaging. I can supply a photo of the cables tomorrow.

chollan, 2018-04-03:
Thanks again @hzeller for all your help!  Here's some photos of my power setup with a photo of the brick that I ordered. i've also added some photos of the rpi before the hat arrives.

https://imgur.com/a/bDyrL

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828