hzeller/rpi-rgb-led-matrix #578: Increase brightness of HUB08 LED panel

DebaGracias, 2018-04-02:
I ordered the led from [AliExpress ](https://www.aliexpress.com/item/40-inch-101cm-ios-And-Android-Wifi-wireless-remote-Programmable-Advertising-LED-Display-Board-Bright-Red/32752577115.html?spm=a2g0s.9042311.0.0.F3UzUv).I was successfully able to use this library to display scrolling text.But I found that the brightness of the LED panel(HUB08) is less. In order to increase the brightness I bought an Adafruit HAT(https://www.adafruit.com/product/2345) . The HAT supports only HUB75 LED boards.
How can I modify this library so that I can use the HAT to increase the brightness?In other words how to map the HUB75 to HUB08 pins?
I know by default the brightness is set to MAX.Is there anything else I can do to increase the brightness?

hzeller, 2018-04-02:
I have not played with HUB08 panels, so I can't give a good answer. As far as I know, the connector is different and they only have a single color ?

I have also heard from people that the output-enable signal needs to be inverted for some single-color displays (but that was mostly HUB12). If the panels don't have an output-enable signal, then setting `--led-pwm-lsb-nanoseconds` to a higher value than 130 and maybe set `--led-pwm-bits=1` will help to increase brightness because you anyway can only show a single color.

 If you already get some output, the Adafruit HAT will probably not make it brighter; it only adapts the logic voltage levels so that they are in-spec.

(Given that it is only one color, you could actually connect up to 9 parallel chains if you want to wire things  up manually and modify SetPixel() accordingly).

DebaGracias, 2018-04-03:
Thank you very much for the quick reply.As mentioned by you I was able to increase the brightness to some extend  by setting --led-pwm-lsb-nanoseconds=50.
Will there be any side effects with this change?Currently it is working fine

hzeller, 2018-04-03:
This should be fine.

Deba22, 2018-04-07:
Hi Hzeller,
We started facing problems when we set--led-pwm-lsb-nanoseconds=50.The raspberry pi screen shows black on boot and gets stuck.If we give some delay say 20 sec before launching the script then the Ras pi screen loads but wifi hangs.We are not able to change the wifi network to connect to wifi and after some time the task bar also hangs.As a result we have to manually remove the power supply and plug it again.
What could be the the reason that the GUI(n WIFI)  of raspberry pi hangs?What is the best value we can set for --led-pwm-lsb-nanoseconds? 
Is there any other way we can increase the brightness?

Deba22, 2018-04-09:
Just to add to my previous comment,if I set --led-pwm-lsb-nanoseconds=75 then the Raspberry Pi GUI does not hang.However I see that the scrolling text on the LED board lags i.e it does not scroll smoothly.What could be the reason?
I have also noticed that if I disable the GUI,the brightness increases slightly.

hzeller, 2018-04-09:
_Never_ use the GUI on the Pi. Always use a headless pi which you connect to via SSH. The Pi is not powerful enough to run a GUI (no matter what the Raspberry Pi foundation wants to tell you).

The nanoseconds is very low, it should work fine even if you increase it to 300 or so. If it gets darker, you need to figure out what the protocol of the Hub08 is (since you have the panel, you can do that)

Deba22, 2018-04-12:
Thank you for your advice.After setting isolcpus=3 the text started scrolling smoothly.

closed 2025-11-27 by marcmerlin