hzeller/rpi-rgb-led-matrix #584: Running picture in rgb led matrix with raspberry pi3

aseelbanijama, 2018-04-09:
Hello everyone... i have rgb led matrix panel 16x32 p6 connect with raspberry pi3 ...can you help me give me a python code that run a picture.ppm?😊

hzeller, 2018-04-09:
If you read the available example code and have specific questions, you can ask them here.

closed 2018-04-09 by hzeller

aseelbanijama, 2018-04-09:
Yes I have a problem ...when I am applying a python code in this link https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/experimental-python-code
The rgb led matrix work very good and rectangular  appear clearly ...but the issue is when run a c-code in this link 
https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/testing  ...
The rgb led matrix doesn't show any responses  (there's no light from any led )???Although I am change the code in example-api-use and use demo code but no response...  I will be thankful if you help me😊😊😊

maddogs, 2018-04-09:
I would suggest that you abandon the Adafruit code and start fresh using this current repository from here, following the install and setup followed by the python examples.

aseelbanijama, 2018-04-09:
Thank you maddogs ... The problem is that I am new to learning the Python language and we must complete the project as soon as ,we try to solve the problem with the language of the C but this did not work🤔

hzeller, 2018-04-09:
We can't help you solve your homework :) Learning a programming language take a while, and you should spend the time doing it.

aseelbanijama, 2018-04-09:
OK thanks very much ☺

angelogoncalve, 2018-04-09:
Do this:
import os
from PIL import ImageFont, Image, ImageDraw
text = (("Raspberry Pi ", (255, 0, 0)), ("and ", (0, 255, 0)), ("Adafruit", (0, 0, 255)))
font = ImageFont.truetype("/usr/share/fonts/truetype/freefont/FreeSans.ttf", 16)
all_text = ""
for text_color_pair in text:
   t = text_color_pair[0]
   all_text = all_text + t
print(all_text)
width, ignore = font.getsize(all_text)
print(width)
im = Image.new("RGB", (width + 30, 16), "black")
draw = ImageDraw.Draw(im)
x = 0;
for text_color_pair in text:
   t = text_color_pair[0]
   c = text_color_pair[1]
   print("t=" + t + " " + str(c) + " " + str(x))
   draw.text((x, 0), t, c, font=font)
   x = x + font.getsize(t)[0]
im.save("test.ppm")
os.system("./demo 1 test.ppm")

aseelbanijama, 2018-04-09:
Thanks angelogoncalve☺^^