hzeller/rpi-rgb-led-matrix #586: how to use a soundcard to allow for this

Wqrld, 2018-04-10:
[image]
A bit unrelated to this project but i got myself an i2s soundcard but how do i make my raspberry use it? I know bmc2835 needs to be disabled as a kernel module but how do i make my raspberry pi output to i2s?

davidlrnt, 2018-04-10:
http://masterandrey.com/posts/en/dac_raspberry_pi_pcm5102/

Wqrld, 2018-04-10:
@davidlrnt thats the first one coming up on google indeed and it tells you how to wire it up but nothing about how to make the raspberry use it software-side

Wqrld, 2018-04-21:
bump, does anyone know how to do this? which kernel modules are needed?

hzeller, 2018-04-21:
This doesn't sound like a question for this forum: you are looking for a driver for your sound-card, which is not really related to showing images on LEDs.
As far as I know, the Raspberry Pi has no i2s output. But it is probably easier if you look in the forums related to to the device you got.

Wqrld, 2018-04-21:
i thought this would be the best place to ask as most of the people here would be using a soundcard. My soundcard is a cheap pmc5102a board from china so it doesnt have a forum

hzeller, 2018-04-21:
Typically, people use $2 USB sound cards that will just work on Linux out of the box.

You have a i2c converter chip which is typically chosen if you have a hardware that outputs i2c audio signal; the Pi does not have such an interface. The only way you can do the same thing with bit-banging on a Pi, but if you already have the LED panel that essentially needs the full GPIO bandwidth, this is probably a challenge.

willemm, 2018-08-10:
The RPi (at least the Pi Zero) certainly does support I2S boards with a hifiberry-dac driver, there are even pihats with I2S chips, for example: https://www.adafruit.com/product/3016
Unfortunately, the I2S interface uses GPIO18 (pin12) which the ledmatrix code also uses for hardware pulsing
I think it should be possible to send PWM0 to a different pin (GPIO12 on pin 32) so that GPIO18 can be used for the soundcard.

hzeller, 2018-08-10:
Yes, the LED matrix uses GPIO18 for its timing sensitive output, but you can change that (at the expense of slightly more flicker).
In the [lib/hardware-mapping.c](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c) file, you can choose the GPIOs you wish to use for the LED matrix / change the ones to leave out for other use (e.g. I²S). The file should be fairly self-explanatory.
If you do change the mapping and used a PCB before, you of course have to manually rewire the changed pins.

willemm, 2018-08-10:
Actually I managed to use GPIO12 on pin32 with hardware pulsing (so no flicker!) by changing SetGPIOMode(gpioReg, 18, 2) to SetGPIOMode(gpioReg, 12, 4)  (which is alternative 0), and adding 12 to the CanHandle function.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828