hzeller/rpi-rgb-led-matrix #587: Help with P10 outdoor panel

CDR114, 2018-04-11:
I have the software working with some indoor P10 panels, but the project I am working on needs the brightness of the P10 16 x 32 outdoor panels.  Unfortunately I haven't had any success in getting the ones I have to work.  Connecting one panel the best result I get is using the command
sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-multiplexing=3

the results are:

[image]

The picture may look strange because the outdoor panel is lying on the indoor panels

Any suggestions?

Can anyone provide a source for a P10 16 x 32 panel that works?

CDR114, 2018-04-12:
I have already tried all the variations of multiplexing and address types and --led-multiplexing=3 and --led-addr-type=0 is the closest to what it should be.  I am going to try individual pixel addressing and see what I get.  But after that I am not sure what to do with that information.

angelogoncalve, 2018-04-12:
Try the last option:

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-multiplexing=7 --led-row-addr-type=0

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-multiplexing=7  --led-row-addr-type=2

[multiplex-mappers.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/1904433/multiplex-mappers.zip)

CDR114, 2018-04-14:
That did not work.  I ran the Pixel Test program with no multiplexing and default address type and for each pixel addressed I get 4 pixels lighted on the panel!

[image]

I am wondering if I have an addressing problem with the Hub75 connector.  My panels have:

R1, B1, R2, B2, A, (blank), CLK, OE down one side
G1, (blank), G2, (blank), B, (blank), STB, (blank) down the other side

The panel uses ICN2028AP Serial Interfaced LED Driver IC's and SM74HC1381 3 to 1 Line Decoder, Multiplexer, Inverting IC and SM24578 and SM24579 IC's which I could not find a datasheet for

CDR114, 2018-04-20:
I just determined from the number of ICN2028AP LED driver IC's that these panels are 1/2 scan  versus the 1/4 scan that I ordered.

angelogoncalve, 2018-04-20:
See the issue 592.

angelogoncalve, 2018-04-22:
Try the option:

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-multiplexing=7 --led-row-addr-type=0

nictruong, 2018-04-22:
@CDR114 I am getting the EXACT same output for `sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-multiplexing=3`. Have you found a solution to your problem yet?

CDR114, 2018-04-23:
I have tried the solution in #592 and I get 8 rows to display properly but get "Error in PixelMapper" for rows 0 through 3 and 8 through 16.  You will notice that 4 rows do not have any error, but at 1/2 scan that equates to 8 displayed rows. Hopefully Alvaro, who developed the solution, can help.  I will try to post a picture tomorrow (It is late and need some sleep).

nictruong, 2018-04-23:
@CDR114, there has been a new commit to the project, --led-multiplexing=7 that has solved all my problems. Hopefuly it will solve yours.

CDR114, 2018-04-27:
The solution in #592 Understanding Multiplexing solved my problem after identifying a defective P10 display panel

closed 2018-04-27 by CDR114