hzeller/rpi-rgb-led-matrix #589: Demos not demoing

Bobarian, 2018-04-18:
I recently purchased a Raspberry Pi 2, Adafruit Hat, and two 32x64 LED Matrix panels (among all the cables and power supply needed to hook up everything).

I followed the guide here: [Raspberry Pi Led Matrix Display](https://learn.adafruit.com/raspberry-pi-led-matrix-display)

With with this I was able to ensure everything worked correctly. I was able to show the numbers on the display, change the matrix.cfg file to orientate things correctly, and it properly duplicates what is on my screen to the LED panels.

However,  I wanted to build this as a piece of art that I can keep plugged in and throw pictures on it.  So I found this guide which was much closer to what I want to do: [Raspberry Pi RGB LED Matrix WebApp](https://learn.adafruit.com/raspberry-pi-rgb-led-matrix-webapp/overview)

This is where I start to run into problems.  I follow all the steps without any issues,  however when I run the demos - it shows it running on monitor but nothing happens with the LED panels.

This is when I found your GIT Hub which it uses and followed this to see if maybe that previous walk through was using an old branch or something.  However walking through everything again,  I'm having the same issue.

I've read through the Read.me files,  I've tried different options such as --led-gpio-mapping=adafruit-hat and --led-slowdown-gpio=2  but nothing seems to work.

I thought that maybe it is because I have more pixels than the the default size (running at 32x32).  And this is when I ran into another issue.  I can change the rows with --led-rows=64 but when I try to change the columns using --led-cols=64 it gives me an error saying that command is unknown (it also doesn't show in the list of commands when I just run demo alone).  Not sure this is really the issue but just thought I'd note it.

So now I'm coming to you as I don't seem to be finding the answer anywhere else.  I'm very very new to all this, and have only done a little tiny bit of javascript programing in my past, so I'm pretty much over my head here.  Thank you in advance for any help you can give.

Bobarian, 2018-04-18:
Just to note, I'm looking to stack the LED panels in a 64x64 square to do a digital picture frame of sorts.

hzeller, 2018-04-18:
If you are using the Adafruit Hat, then using the `--led-gpio-mapping=adafruit-hat` flag should work.
You have not specified with 'nothing seems to work' means.
Did you run the demo program in the `examples-api-use/` directory ?

If it says that it doesn't find the program: remember if you are running the program, you need to give it the path where it is, as it otherwise is of course not found. So call
```
 ./demo
```
instead of 
```
demo
```
(if you are in the examples-api-use/ directory already. if not, then `examples-api-use/demo`)

marauder37, 2018-04-19:
The tutorial you're following is using an old fork of the project which does not have `--led-cols`. git blame says that `--led-cols` was added on 2018-01-20 which is quite recent. There is no longer a reason to use the old fork, so check that you have the latest version of this project.

If a command isn't working, copy and paste the command you ran and the output you got. Otherwise we're just guessing about what happened.

Bobarian, 2018-04-19:
@hzeller -  Sorry,  when I meant 'nothing seems to work' this was me trying to run various demos and using the different flags (example: sudo ./demo --led-gpio-mapping=adafruit-hat --led-rows=64 --led-slowdown-gpio=2).  Even when I ran this,  the Pi would look like it is working (showing on monitor) but the LEDs never turn on.

@marauder37 - THANK YOU!!  You're mention of --led-cols not being added made me wonder that perhaps even though I had pulled this fork,  that it was possibly still using old files.  So I went through and did a full wipe of all the directories,  then cloned this repository again (doing this first before following any other instructions) and this got me up and running.  I am now able to run the demos and they work on the LEDs.

Now my next step - figuring out how to set the LED panels into the right order.

Thank you for the answers and help.  Sometimes it just needs a little nudge that sparks the answers. :)

closed 2018-04-19 by Bobarian