hzeller/rpi-rgb-led-matrix #590: Examples not showing

ashleebeggs, 2018-04-18:
Hi, I have one 32x16 panel with raspberry pi 3 and adafruit RGB RTC hat. I followed the instructions and tried running demo 1 and 3. Both show the panel split in half with blue and purple. 

I'm not sure where to look for the issue. 
I'm not getting any errors either

Is there something I'm missing?
Thanks! 
 
[image]

ashleebeggs, 2018-04-18:
I run sudo ./demo -D 3 --led-gpio-mapping=adafuit-hat --led-rows=16

angelogoncalve, 2018-04-19:
Try this in python:

from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 16
number_of_panels = 1
parallel = 1
number_of_columns = 32

def rgbmatrix_options():
    options = RGBMatrixOptions()
    options.multiplexing = 0
    options.row_address_type = 0
    options.brightness = 100
    options.rows = number_of_rows
    options.cols = number_of_columns
    options.chain_length = number_of_panels
    options.parallel = parallel
    options.hardware_mapping = 'adafruit-hat'
    options.inverse_colors = False
    options.led_rgb_sequence = "RGB"
    options.gpio_slowdown = 4
    options.pwm_lsb_nanoseconds = 130
    options.show_refresh_rate = 0
    options.disable_hardware_pulsing = True
    options.scan_mode = 1
    options.pwm_bits = 1
    options.daemon = 0
    options.drop_privileges = 0
    options.pixel_mapper_config = "U-mapper;Rotate:0"
    options.pwm_dither_bits = 0
    return RGBMatrix(options=options)

display = rgbmatrix_options()

height = display.height
width = display.width

print ('width={}, height={}'.format(width, height))

class Point:
    def __init__(self, x, y):
        self.x = x
        self.y = y

    def __repr__(self):
        return '({}, {})'.format(self.x, self.y)

for x in range(0, width):
    for y in range(0, height):
        display.SetPixel(x, y, 0, 255, 0)
        point = Point(x, y)
        print point
        time.sleep(1)

ashleebeggs, 2018-04-19:
Thanks for the response! 
I'm new to the pi and I'm not quite sure where to put the code. Is there an existing file I should add or edit this to? 

Thanks

angelogoncalve, 2018-04-19:
Yes, first create a .py file name and put the code with the correct indentation, use for example a python editor named PyCharm software that runs in Windows operating system computer, or you can use the Python2 compiler that comes in Debian operating system from raspberry.

Then in the command line use the command:

sudo python name_file.py

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin