hzeller/rpi-rgb-led-matrix #593: P10 HUB12 Multiplexing Almost There

cmagnuson, 2018-04-19:
I'm trying to get a P10 panel with HUB12 interface working and am almost there, but can't get the mapping quite right.

I'm using the adafruit matrix HAT and rewiring the HUB75 OE/GND/A/B/CLK/LAT/R to matching HUB12 pins seems to be working correctly.

The problem I'm seeing is that with the following mapping, I get the same results duplicated on left half and right half of display:
```
class P10Mapper : public MultiplexMapperBase {
public:
  P10Mapper() : MultiplexMapperBase("P10", 2) {}

  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    int y_component_matrix_x = ((int)(y/4)) * 8;
    int x_component_matrix_x = (((int)(x/8)) * 32) + (x % 8);
    *matrix_x = x_component_matrix_x + y_component_matrix_x;
    *matrix_y = 0 - ((y % 4) - 3);
  }
};
```
Using this command: 
`sudo ./demo -D12 --led-gpio-mapping=adafruit-hat --led-rows=16 --led-multiplexing=7
`
[image]

And with another mapping I can solve the left-right split, but then only every 2 rows are displayed (2 on 2 off, etc):
```
class P10Mapper2 : public MultiplexMapperBase {
public:
  P10Mapper2() : MultiplexMapperBase("P10-2", 4) {}

  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
    int y_component_matrix_x = ((int)(y/4)) * 8;
    int x_component_matrix_x = (((int)(x/8)) * 32) + (x % 8);
    *matrix_x = x_component_matrix_x + y_component_matrix_x;
    *matrix_y = 0 - ((y % 2) - 1);
  }
};
```
`sudo ./demo -D12 --led-gpio-mapping=adafruit-hat --led-rows=16 --led-multiplexing=8 --led-chain=1`
[image]

I've experimented with the led-row-addr-type and the existing multiplexers but with no success.  Any ideas would be greatly appreciated!

cmagnuson, 2018-04-23:
For what it's worth, I was able to map out each pixel with the following command:
`sudo ./demo -D12 --led-gpio-mapping=adafruit-hat --led-rows=16 --led-cols=32 --led-chain=4
`

When I did less then led-chain=4, I was getting duplicate pixels (display split left and right or top and bottom).  I created the following x/y mapping for them, but haven't been able to successfully implement a multiplex mapper that works sized for the 16x32 display

X-mapping:
[image]

Y-mapping:
[image]

I also tried a command that should be appropriately sized, but after filling rows 0 and 1, rows 2 and 3 did not fill:
` sudo ./demo -D12 --led-gpio-mapping=adafruit-hat --led-rows=4 --led-cols=128 --led-chain=1`

NeerajShrivastava04, 2018-06-02:
did you get a solution for your issue?

cmagnuson, 2018-06-04:
I didn't but thanks for checking back. I switched to some RGB panels that were supported by rpi-rgb-led-matrix and haven't done any more experimenting with these P10 panels since.  Issue can be closed in my opinion

NeerajShrivastava04, 2018-06-08:
I have written a transformer for my single color P10 module.
Please see #531 .
May be it will be useful to you.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828