hzeller/rpi-rgb-led-matrix #597: No LED being turned on

nictruong, 2018-04-21:
I bought a 16x32 matrix from Aliexpress: https://www.aliexpress.com/item/Freeshipping-outdoor-p10-smd-led-display-module-16x32-32x16-160x320-320x160-1-2-scan-high-brightenss/32669606028.html

I wired and rewired according to the README and I ran this command:

`sudo examples-api-use/demo -D0`

This is what I see: 

`FYI: Couldn't set affinity 0x8: Invalid argument

Size: 32x16. Hardware gpio mapping: regular

Press <CTRL-C> to exit and reset LEDs`

But not a single LED is being turned on. 

[image]

In the README, there is a 'c' pin, which I don't see. did I rightly assume it is the same as 'NC'? I have 3 wires not connected to anything: the 3 bottom GND pins of the right column.

This is an overview of what I have:

[image]

Can someone point me in the right direction? `FYI: Couldn't set affinity 0x8: Invalid argument` how can I fix this message? Is my matrix defective?

I am a JavaScript dev, I was going to play around with the NodeJS bindings for this, but that project is not maintained anymore.

nictruong, 2018-04-22:
Thank you for your quick reply.

I was trying a bunch of commands with different values of --led-multiplexing and --led-row-addr-type, got frustrated because none of them worked, and decided to switch the 2.1mm plug jack adaptor connecting to the power supply, I now see a board with turned on LEDs!

I really appreciate you trying to help me. Cheers.

closed 2018-04-22 by nictruong

hzeller, 2018-04-22:
Ha, yes, sometimes things as simple as the power supply can be the culprit :)

angelogoncalve, 2018-04-22:
See the issue https://github.com/hzeller/rpi-rgb-led-matrix/issues/592.

Try the option:

sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-multiplexing=7 --led-row-addr-type=0

nictruong, 2018-04-22:
Hi angelogoncalve

I tried your command (after downloading and compiling the multiplex-mappers.cc file containing --led-plexing=7). I get a garbage image:

[image]

I have tried every combinations of the following commands:

` sudo ./demo -D3 --led-rows=16 --led-cols=32 --led-multiplexing={0-7} --led-row-addr-type={0-2}`

Reading the README, since I have a A-B panel, it would seem I should be using `--led-row-addr-type=1`, but alas, it has not given me good results. `-D3` is supposed to give me a square right? Do you have any other advice for me?

nictruong, 2018-04-22:
Just saw that there was a new PR probably merged after I cloned the project. Will try that and report back.